wellenvogel/avnav #311: AIS ship data show wrong ship sizes

Scorgan01, 2023-09-25:
I observed wrong ship size data (i.e. length, beam) in AIS ship data entries. As far as I can see, size data for B-class messages is correct, but data of A-class messages is wrong (e.g. length=1022m, beam=126m, correct data l=31m, b=6m). All other field entries seem to be correct.
I get the AIS sentences via N2K from an B&G V50 VHF where data is shown correctly. I receive the data on AvNav via WiFi gateway esp32-nmea2000. So, I cannot tell if the error is on the gateway or in AvNav. I decided to create the issue here, though. The good thing is that you are maintainer of both solutions.

wellenvogel, 2023-09-25:
Do you have some sample data from the n2k bus?

wellenvogel, 2023-09-26:
You could collect it from the USB port of the ESP32.
Set it to actisense mode and connect it to a computer with a terminal program that can write the log data to a file.
E.g. Putty.

Scorgan01, 2023-09-26:
All right. Let me collect some sample data.
I'll be only back at the boat in about two weeks and will try to get the data, then.

wellenvogel, 2023-09-29:
I just make some small changes in the gateway code.
If you would like to test, just contact me and I could send you an updated code for the gateway to test.

Scorgan01, 2023-10-02:
I'll be on the boat again in two weeks. I'll be glad to test the code, then. Can I download the code here or do you have to send it to my mail address?

wellenvogel, 2023-10-02:
As there is no new release it would have to send it somehow.

Scorgan01, 2023-10-15:
Cannot reproduce error with new test version of gateway code anymore. Error seems to be fixed.
Thank you.

closed 2023-10-15 by Scorgan01