wellenvogel/avnav #313: Bug when saving settings

AaskovKnudsen, 2023-11-09:
Hi Andreas
I think there is a small bug in avnav version 20230705
If I chose "speed(kn)" as second AIS label and try to save this setting, I get an error saying "speed is not a valid value for aisSecondLabel"
Please see attached screenshots.
Best

[image]
[image]
[image]

wellenvogel, 2023-11-11:
Thanks for bringing this up.
Can reproduce it.
It's a bug in the checking code when saving the settings.

wellenvogel, 2023-11-11:
daily: https://www.wellenvogel.net/software/avnav/downloads/daily/20231111/

wellenvogel, 2024-05-20:
20240520

closed 2024-05-20 by wellenvogel