wellenvogel/avnav #315: AvNav not process my gemf charts collection on dragging mouse

MENIER, 2024-01-13:
Hello! i"m hanging arround from month on this project. Yes it's coming work!
1)I Put my mbtiles files charts in the directory: **avnav/import/**
2)Avnav  convert automatically them in gemf in the directory: **avnav/charts/**
3)Load one of my maps ok! I view the map perfectly.
4)Drag zoom etc... doesn't upload the next best map automatically
It is a normal running way of AvNav?

wellenvogel, 2024-01-13:
There is no need to convert mbtiles.
Just upload them inside the app.
What is displayed when you are zooming heavily depends on the content of the mbtiles - i.e. wich zoom levels are included.
Normally I would recommend to have at least every second zoom level being packed into the mbtiles.
AvNav will try to load and zoom up lower zoom levels if the requested one is not found - by default up to 4.

MENIER, 2024-01-13:
When I"m zooming heavily i see max zoom of the unique map (Metadata:  ZOOM_LEVEL=8)  i have selected.
I have not  upload  my maps inside the app. I have just copy maps in the directory: avnav/charts/ 
Avnav GUI see the list of all my maps and can show alls but not automatically in drag zoom mode....
Is there a chartlist.dat like in opencpn?
I have tried Creating a int@21603.mbtiles.cfg with it I see 3 differents maps on gui:
```
{
  "useDefault": true,
  "name": "int@21603.mbtiles.cfg",
  "overlays": [
    {
      "chartKey": "int@400.mbtiles",
      "type": "chart",
      "opacity": 1,
      "enabled": true,
      "bucket": "L2",
      "overlayId": 3
    },
    {
      "chartKey": "int@21602.mbtiles",
      "type": "chart",
      "opacity": 1,
      "enabled": true,
      "bucket": "H1",
      "overlayId": 1
    },
    {
      "chartKey": "int@21604.mbtiles",
      "type": "chart",
      "opacity": 1,
      "enabled": true,
      "bucket": "H1",
      "overlayId": 2
    }
  ],
  "defaultsOverride": {}
}
```

wellenvogel, 2024-01-13:
copying into the chart dir is ok,
But you should not try to generate overlay configs by hand.
There is a couple of things that those configs have to respect to function.
Not documented at all.
Just use the Webapp for that.
But overlays will not solve your problem.
So it would be good to know which zoom levels are in your mbtiles.
If you have different mbtiles for different zoomlevels ( makes no real sense...) - just merge them into one mbtiles db before using it in AvNav.

MENIER, 2024-01-13:
I made my mbtiles from different BSB  charts like that example:
```
gdal_translate -co "ZLEVEL=9" -of mbtiles 	21602a.nos	21602a.mbtile
gdaladdo -r nearest	21602a.mbtiles
```
Merge all my mbtiles charts in one this is the solution? No problem of memory on Rasperry?

free-x, 2024-01-13:
For KAP files there is a better way
Copy your KAP files for example to ~/tmp
Then
/usr/lib/avnav/chartconvert/read_charts.py -g -m all -o MYGEMF tmp/
After coffeebreak you can find resulted gemf in ~/AvNavCharts/out/
Then you copy/move gemf to ~/avnav/data/charts
Good luck

MENIER, 2024-01-14:
After not coffeebreak but champagneBreak! Thanks a lot it's run perfectly. Next step connect a GPS,TCP etc.. GEMF is proprietary format?

free-x, 2024-01-14:
> GEMF is proprietary format?
https://www.cgtk.co.uk/gemf

wellenvogel, 2024-01-14:
It has some advantages over mbtiles - simple binary format with readers being available in nearly any language.

wellenvogel, 2024-01-14:
Closing this as it's no real issue.

closed 2024-01-14 by wellenvogel