wellenvogel/avnav #318: demo server charts, BSH charts not working

quantenschaum, 2024-01-15:
Thank you for this great project!

The charts on the demo server provided by the BSH do not work anymore because the layers in the WMS have changed.

I have been looking into the GeoSeaPortal WMS and tried to create charts for [OsmAnd](https://osmand.net/) and [OSM](https://www.openstreetmap.org) from this data. Maybe you are interested what I have done and can use it somehow.

Have a look at https://github.com/quantenschaum/mapping and http://waddenzee.duckdns.org

wellenvogel, 2024-01-15:
Many thanks, that looks really cool.
Will have a much closer look on it.
Unfortuantely BSH dropped the soundings from their Seaportal layers (by my feeling they violate some EU regulations with this - but they did this intentionally as tehy told us on a fair when we talked to them).
But using the geojson an raster from there is a great approach.

wellenvogel, 2024-01-15:
One more question:
Would it be ok to use your server to provide the charts for the demo?
And I could also imagine to add some mapproxy setting so that users could download charts inside AvNav. But I'm not sure about the traffic that it will create...

quantenschaum, 2024-01-22:
FYI Here are some map sources that can be used in mapproxy. Using them in AvNav does probably to agree with their terms of use. But it's maybe good to know anyways.

The way you integrated mapproxy is very nice, especially the predefined bounding boxes for different zoom levels for the seeding for offline usage. And the mbtiles can be downloaded and copied to other devices. :+1: 

BTW The mtbiles can be [easily converted](https://github.com/quantenschaum/mapping/blob/master/mbtiles2osmand.py) to sqlite for use in OsmAnd.

```yaml
sources:
  s_no-sjokart:
    type: tile
    grid: osm_grid_xyz
    transparent: false
    # https://www.norgeskart.no/?test=1#!?project=norgeskart&layers=1008
    url: https://cache.kartverket.no/sjokartraster/v1/wmts/1.0.0/?layer=sjokartraster&style=default&tilematrixset=googlemaps&Service=WMTS&Request=GetTile&Version=1.0.0&Format=image/png&TileMatrix=%(z)s&TileCol=%(x)s&TileRow=%(y)s

  s_fr-shom:
    type: tile
    grid: osm_grid_xyz
    transparent: false
    # https://data.shom.fr/
    url: https://services.data.shom.fr/clevisu/wmts?layer=RASTER_MARINE_3857_WMTS&style=normal&tilematrixset=3857&Service=WMTS&Request=GetTile&Version=1.0.0&Format=image/png&TileMatrix=%(z)s&TileCol=%(x)s&TileRow=%(y)s
    http:
      headers:
        Referer: https://data.shom.fr/
        User-Agent: Mozilla/5.0 AppleWebKit/537.36 Chrome/120.0.0.0 Safari/537.36

  s_se-sjokort:
    type: wms
    # https://geokatalog.sjofartsverket.se/kartvisarefyren/
    req:
      url: https://geokatalog.sjofartsverket.se/mapservice/wms.axd/FyrenBakgrund
      layers: OgcWmsLayer0
      transparent: true

  s_nl-rws:
    type: wms
    req:
      url: https://geo.rijkswaterstaat.nl/arcgis/rest/services/ENC/mcs_inland/MapServer/exts/MaritimeChartService/WMSServer
      layers: 0,2,3,4,5,6,7
      transparent: true

  s_nl-noaa:
    type: wms
    req:
      url: https://gis.charttools.noaa.gov/arcgis/rest/services/MACHC/MACHC_ENCOnline_Public/MapServer/exts/MaritimeChartService/WMSServer
      layers: 2,3,4,5,6,7
      transparent: true

  s_cmapx:
    type: tile
    grid: osm_grid_xyz
    transparent: false
    # https://www.c-map.com/de-de/chartexplorer/
    url: https://tiles.c-map.com/wmts/cmt_int1/webmercator/%(tms_path)s.png
    http:
      headers:
        Referer: https://www.c-map.com/
        User-Agent: Mozilla/5.0 AppleWebKit/537.36 Chrome/120.0.0.0 Safari/537.36

  s_oceanwise:
    type: wms
    req:
      url: https://maps.oceanwise.eu/data/enc/
      layers: baseWater,hydrography,hydrographyLabel,soundings,dangers,platforms,pipelines,cables,restrictions,anchors,aton,atonLabel,baseLabel
      transparent: true
    http:
      headers:
        Referer: https://maps.oceanwise.eu/
        User-Agent: Mozilla/5.0 AppleWebKit/537.36 Chrome/120.0.0.0 Safari/537.36

  s_osmand:
    type: tile
    grid: osm_grid_xyz
    transparent: false
    # https://osmand.net/map
    url: https://maptile.osmand.net/tile/nautical/%(tms_path)s.png

  s_world-imagery:
    type: tile
    grid: osm_grid_xyz
    transparent: false
    url: https://server.arcgisonline.com/arcgis/rest/services/World_Imagery/MapServer/tile/%(z)s/%(y)s/%(x)s

  s_marine-traffic-density:
    type: tile
    grid: osm_grid_xyz
    transparent: true
    url: https://tiles3.marinetraffic.com/densitymaps2022/0/%(tms_path)s.png

  s_nz-wetmaps:
    type: tile
    grid: osm_grid_xyz
    transparent: true
    url: https://tiles.wetmaps.co.nz/file/wetmaps-tiles/sync/tiles/%(tms_path)s.png

  s_us-noaa:
    type: wms
    req:
      #url: https://geoint.nrlssc.navy.mil/dnc/wms/DNC-WORLD/basemap/DNC-BASEMAP
      #layers: DNC-BASEMAP
      #url: https://geoint.nrlssc.navy.mil/enc/wms/ENC/basemap/ENC-BASEMAP
      #layers: ENC-BASEMAP
      url: https://gis.charttools.noaa.gov/arcgis/rest/services/MCS/ENCOnline/MapServer/exts/MaritimeChartService/WMSServer
      layers: 6,5,4,3,2,1,0
      transparent: true

  s_ca-enc:
    type: wms
    req:
      url: https://e-nav.ccg-gcc.gc.ca/geoserver/enav/enc_transparent/ows
      layers: enc_transparent
      transparent: true

  s_uk-ho:
    type: tile
    grid: osm_grid_xyz
    transparent: true
    # https://www.visitmyharbour.com/
    url: https://media.visitmyharbour.com/0.001-new/defaced-uk/%(tms_path)s.png
    http:
      headers:
        Referer: https://www.visitmyharbour.com/
        User-Agent: Mozilla/5.0 AppleWebKit/537.36 Chrome/120.0.0.0 Safari/537.36

  s_eniro:
    type: tile
    grid: osm_grid_tms
    transparent: false
    # https://kartor.eniro.se/s%C3%B6k/sj%C3%B6kort
    url: http://map01.eniro.com/geowebcache/service/tms1.0.0/nautical/%(tms_path)s.png
    http:
      headers:
        Referer: https://kartor.eniro.se/
        User-Agent: Mozilla/5.0 AppleWebKit/537.36 Chrome/120.0.0.0 Safari/537.36
```

wellenvogel, 2024-01-22:
Many thanks for providing the sources.
But some of them are not in line with their license terms.
Some time ago the eniro team already contacted me and asked me to remove their source from my demo.
So I have to delete the sources here to avoid any discussions.

quantenschaum, 2024-02-09:
I reworked the charts at http://waddenzee.duckdns.org/ and if you want to use them in the online demo you can grab this file http://waddenzee.duckdns.org/bshx.mbtiles

I also wrote me a NMEA simulator to test and play with AvNav and SignalK at home. Maybe you want to have a look at it or even use it for the for online demo, too. Would make whole thing more interactive.

https://github.com/quantenschaum/mapping/blob/master/ship_sim.py

wellenvogel, 2024-02-11:
Thx.
Not sure about the charts - it's always a bit a licensing issue. I had some discussions with the BSH - and basically using their WMS was ok. But I'm not sure about some mbtiles...

For the simulator:
The online demo cannot handle NMEA data at all. It basically uses some special format to repeat a sequence of position and AIS data.
For NMEA simulation I use the simulator from Timo: https://www.kave.fi/Apps/index.html

quantenschaum, 2024-02-11:
I asked the BSH if I could use data from geoseaportal to import them into openstreetmap. An official answer is still work in progress, but they told me: foreseeably yes, see http://waddenzee.duckdns.org/download/BSH.md

Thanks for the link, didn't know that. It was just an idea.

closed 2024-02-11 by quantenschaum

quantenschaum, 2024-02-11:
You said they had spot soundings available but removed them. Can you tell me more about why?

wellenvogel, 2024-02-11:
Yes, I guess it was in 2020 when we talked to them at the "Boot" in Düsseldorf. And they told us that they intentionally removed them due to their licensing understanding...
Before it looked like:
[image]

quantenschaum, 2024-02-11:
Schade, dass das weg ist. Die wollen sicher nicht, dass man sich seine eigenen Karten bastelt und die schon brav kauft. In NL gibt's ja ENCs zum Download. Naja, ich denke, es passt  schon zur OpenData-Lizenz, wenn ich mir "meine Karte" erzeuge und verteile. Kannst du verwenden, wenn du magst.

free-x, 2024-02-12:
I think I have an idea how to overcome the legal hurdle
According to the MBTiles Spec (https://github.com/mapbox/mbtiles-spec) you can define "attribution" in the metadata table ( HTML string: An attribution string that explains the data sources and/or the style of the map)
And AvNav should then evaluate and display/indicate this accordingly

quantenschaum, 2024-02-12:
The [Geodatennutzungsverordnung](https://www.bsh.de/DE/THEMEN/Geoinformationen/_Anlagen/Downloads/Geonutzv.pdf?__blob=publicationFile) (what a word) requires 

>Veränderungen, Bearbeitungen, neue Gestaltungen oder sonstige Abwandlungen [...], **der beigegebene Quellenvermerk gelöscht wird.**

The rationale for this is: a map derived from BSH data and maybe combined with other data, like in the case of OSM or "my map", has not been officially created by the BSH. The end user should not think that the BSH created this map because it is mentioned as data source and probably make them responsible for anything related to it. So, it is better not to mention the BSH at all as source directly. There could be an explanation somewhere describing the process of creating the map based on their open data though, but there must not be a little attribution box somewhere shown along with the map mentioning BSH.

In the case of OSM it is even _technically impossible_ to add such a "Quellenvermerk" to the data in the database and to enforce the display of it in all possible end uses of OSM data. So, having no source mentioned at all is the best option and this makes it even possible to use this data in OSM, otherwise it would violate the terms of use.

[DWD](https://www.dwd.de/DE/service/copyright/vorlagen_quellenangabe.html) and [WSV](https://wiki.openstreetmap.org/w/images/c/c6/Terms_of_usage_ienc.pdf) handle it like that with their open data products.

- If you use them as they are, unmodified, like an image created by them, you have to _mention the source_.
- If you edit the data, derive own work from it, mix it with other data, so what the end user sees is your work and not theirs, then you _must not mention them as source_ to avoid confusion.

[BSH told me](http://waddenzee.duckdns.org/download/BSH.md) that they will handle it as WSV/DWD, so they must not be mentioned as source of the map.