wellenvogel/avnav #319: Can't adjust longitude

gtf375, 2024-01-29:
AvNav version 20230705 on Raspberry Pi.
In the route editor it is not possible to save longitudes greater than 90 degrees.

wellenvogel, 2024-01-29:
Can you provide some symptoms (screenshots) and a detailed description please.
I just did a short check - no problems.

gtf375, 2024-01-30:
I change the longitude in the route editor and save.
[image]
[image]

The longitude value has not changed.
[image]

If the longitude is from 0 to 90 degrees, everything is saved. If the longitude is from 90 to 180, then it is not saved.

wellenvogel, 2024-02-07:
thanks, I could reproduce the bug.
Correction will be available soon.

gtf375, 2024-02-07:
Hello Andres. Thank you. I also have a desire
enter the latitude on the first line and longitude on the second line in the Edit Waypoint window, as on all commercial plotters.

wellenvogel, 2024-05-20:
20240520

closed 2024-05-20 by wellenvogel