wellenvogel/avnav #320: true wind mapping from signalk

quantenschaum, 2024-02-05:
In [the docs](self.api.getDataByPrefix('gps')) your write how true wind is mapped

- `environment.wind.speedTrue / environment.wind.speedOverGround --> gps.trueWindSpeed`
- `environment.wind.directionTrue / environment.wind.directionGround --> gps.trueWindAngle`

You have to be careful here. True wind usually is the wind relative to *water* not ground! This is what sailors mean talking true wind. The weather report gives you ground wind. The difference comes from water current/tide.

Why? - Because the wind relative to the water is the wind a sailboat can sail with. Imagine ground wind from east and your boat is in the water with a west going current same speed as the wind. The true wind (water wind) in this case is zero and you cannot sail. If there was zero ground wind in this situation, you would experience a wind from the west when moving with the tide and you can use this to sail north through the water. Of course in both cases you move over ground due to the current.

I am currently looking into code from [signalk](https://github.com/SignalK/signalk-derived-data/) and the [sail instrument](https://github.com/kdschmidt1/Sail_Instrument/issues/9). Both get the calculations of wind and current wrong.

quantenschaum, 2024-02-05:
Is there actually a list of all `gps.*` fields with an explanation what they mean?

wellenvogel, 2024-02-06:
I agree with your comments.
On the other hand I try to leave those decision to SignalK.
But maybe I should go for a further separation between true and ground.
Unfortunately I was not really able to find some "agreed" definitions we could follow so that everyone has the same idea.

For the gps values:
There is no such doc right now.
Additionally plugins can extend this.

quantenschaum, 2024-02-06:
Nice. I think there is no way around having ground wind and true wind separated if you want to get it right.

What happens if SK supplies both `wind.speedTrue` and `wind.speedOverGround`?

Of course, plugins can define their own data fields, but a well defined base set (COG,SOG,HDT,HDM,AWA,TWA,GWA,...) would be nice to have. It seems to exist somehow, but I am missing ground wind, set and drift, leeway, wind directions.

I was working on the [Sail_Instrument](https://github.com/kdschmidt1/Sail_Instrument) and tried to set `gps.trueWindAngle` and `gps.trueWindSpeed`. This seems impossible: `setting invalid path gps.trueWindAngle` (was registered in `pluginInfo()` with `{"path": "gps.trueWindAngle", "description": "true wind angle"}`)?

And why are some widgets smaller that others, DPT vs HDT?

quantenschaum, 2024-02-06:
FYI course data calculations, maybe you can use that somehow.

https://github.com/quantenschaum/mapping/blob/master/coursedata.py

wellenvogel, 2024-02-07:
Thanks for the link.
But what I was looking for is some sort of agreed defintion of the various terms.
This seems to be difficult -  I found a couple of diverting discussions ( and have some understanding from what I learned in the past).
So if you know a source of such agreed definitions this would be helpful.
For the plugin:
Normally AvNav does not allow plugins to overwrite internal gps Values.
You could change this by setting allowKeyOverwrite in the config of the plugin ( only directly in the xml).
For the widgets:
Sizes depend a bit on their primary location in the default layout. But this part is waiting for some restructuring to make things more clear.

wellenvogel, 2024-02-18:
I will not change existing internal definitions if not absolutely necessary. Otherwise it would easily break layouts that people defined.
So I guess the easiest way would be to add new values for groundWindXXX - and maybe make the fallback for the SK handling optional. https://github.com/wellenvogel/avnav/blob/5321248f6be76ebea5ff36b0c746ac5658332df9/server/avnav_nmea.py#L64
And this has to be done for Android too.
For the plugin I always proposed to go for own values. Display widgets for values not known in AvNav are not there any way (you would have to use the default widget). So it would be no problem for the plugin to have other pathes.
Or - if AvNav has a decoder - just send the NMEA. This is the preferred solution any way as other systems that would connect will also be able to get the computed NMEA sentences.

wellenvogel, 2024-02-18:
I just tried to clean this a bit - but as far as I see the SK definitions are really inconsistent. So they have 
https://signalk.org/specification/1.5.0/doc/vesselsBranch.html#vesselsregexpenvironmentwinddirectiontrue
and describe this as direction against true north. But I guess following your proposed definitions this should be directionGround. And the wind direction against HDT (i.e. trough water) is not there at all. Finally I guess I also found some discussion whether this would make sense at all.
Basically what you utilize is -maybe-  the angle through water.
So this makes things more complicated when trying to use SK as source. Until now there is the clear rule that AvNav does not compute derived values by it's own (except for RMB, APB). So what do you think would be the best approach when considering the SK interworking?
And we always need to keep in mind that people now maybe defined their dashboards with the current wind values. And it would be frustrating of they break now.
I agree with your thoughts - but on the other hand for many situations it is just OK if you don't have some heading data in your nmea network and just use COG - and know about the problems and errors that you get.

wellenvogel, 2024-02-20:
Mostly agree.
At least 2 points:
(1)I guess you did not get my point about _sending_ NMEA data. If the plugin is sending it's computed values as NMEA (assuming that AvNav could decode it) instead of writing to the AvNav store they will also be available to other users of the NMEA stream. If you only write it to AvNav's internal store they will only be visible to AvNav.
But this could potentially introduce a looping problem if the NMEA is additionally send to SK and SK reading is enabled (this is a misconfiguration - but could happen...).
So maybe sending NMEA could be optional.

(2) For the definition:
Yes, would make sense to have the internal names documented. But still I would like to keep the existing ones as they are (potentially changing the SK mapping with a fallback option). For new ones we can freely select (your proposals are OK from my pov).

Overall I was really thinking of maybe going away from the strict "AvNav does not compute derived data" towards a set of computations to be included in AvNav itself. Just to get feature parity with the Android part (where we don't have plugins right now).
But this requires to carefully think about the looping problem again.

quantenschaum, 2024-02-20:
>you did not get my point about sending NMEA

You mean an avnav plugin sending NMEA to avnav itself not somewhere else? or that avnav gets data from SK as NMEA not via the SK protocol?

>would make sense to have the internal names documented :+1: 

>I would like to keep the existing ones as they are

No need to change anything, it's fine what there is.

>a set of computations to be included in AvNav itself.

would make sense if avnav is the master :brain: of the setup

My main question here was about definitions of well-known paths in avnav.

wellenvogel, 2024-02-20:
> You mean an avnav plugin sending NMEA to avnav itself not somewhere else? or that avnav gets data from SK as NMEA not via the SK protocol?

Instead of setting values in the store the plugin could also enqueue NMEA sentences for the computed values (as long as AvNav will be able to decode them).
But with the mentioned problems.

wellenvogel, 2024-03-31:
I guess for now a first step will be to omit the SK fallback mappings an just to provide what can be decoded from NMEA sentences:
AWA,AWS,TWA,TWD,TWS
As for TWD there is no clear mapping in SK I would omit any SK mapping for now.

At first this would be inline with the general approach of AvNav not computing additional data.
And AvNav's wind widgets already have the option to select between AWA/AWS or TWA/TWS with an automatic fallback option (but showing what is displayed).
The question is if (and if yes: where) AvNav should compute the missing values. Currently I feel there are 2 places where such computations are executed:

- [SailInstrument Plugin ](https://github.com/kdschmidt1/Sail_Instrument/blob/master/Sail_Instrument/plugin.py)- not writing to the store but generating some missing NMEA sentences - you already incorporated the formulas there
- [more-nmea-plugin](https://github.com/kdschmidt1/avnav-more-nmea-plugin/blob/master/plugin.py) - writing to the store (but with different names) - and I guess at least some different computations.

For Android we have none. So I am wondering if it would make sense to move the computations directly to AvNav but at the display side (i.e. in the JS code).
Advantage: Solution works for all platforms
Drawback: No NMEA sentences with the generated data

Not sure about the best solution yet. But I guess at least we should avoid duplicate + (potentially) different computations in 2 plugins (I already asked KD for the status...).
What do you think?

quantenschaum, 2024-04-01:
>general approach of AvNav not computing additional data

This is not quite true, strictly speaking: [HDT from HDM](https://github.com/wellenvogel/avnav/blob/443428c7a561b86b2b95c8f5cc46671228f33a7c/server/avnav_nmea.py#L535), but I of course know what you mean

>SailInstrument Plugin - not writing to the store but generating some missing NMEA sentences - you already incorporated the formulas there

That is exactly what we (@kdschmidt1) did. We merged the SI with the more nmea plugin, making the latter obsolete since it is redundant. 

Question: How does the Android implementation work? Is it a complete rewrite in Java or in the Python code executed? Why do plugins not work on android?

wellenvogel, 2024-04-02:
> Question: How does the Android implementation work? Is it a complete rewrite in Java or in the Python code executed? Why do plugins not work on android?

Yes - complete rewrite in Java. And does not really has a plugin interface right now.
To a certain degree it provides the same features. Especially I try to provide the same NMEA handling capabilities.

quantenschaum, 2024-04-04:
>complete rewrite in Java

That's tough! Just thinking: Wouldn't it be possible to reuse the existing Python code and run it on Android? Natively or in a Jython VM? With some glue Java code to access the hardware like the GPS or a USB serial adapter?

wellenvogel, 2024-04-04:
Most probably there would be chances to do that.
But I guess there is a lot of stuff that will not be available (all the libs with native parts) - and it could get hard to build and maintain such an environment.
Initially I had a rough look at such an architecture - but dropped it soon.
The more you follow some exotic path the harder things can be maintained in the long run.
At the end one of the bigger challenges is to always keep up with all the changes that google requires - and this needs a lot of java code any way.
Fortunately there are some NMEA and AIS decoding libs available for java.

wellenvogel, 2024-05-20:
20240520

closed 2024-05-20 by wellenvogel