wellenvogel/avnav #322: idea/feature request: add common classical navigation tools

quantenschaum, 2024-02-08:
I just came across [AngelNav](https://www.angelnav.co.uk/) by Tom Cunliffe. It features some nice tools for classical chart work like you would do it on paper charts.

- draw a bearing line
- draw a range circle
- fix from multiple bearings
- calculate course to steer
- estimated position

There are some videos on how this works and looks like. https://www.angelnav.co.uk/videos.html

It would be nice to have this available in AvNav as well.

I made a quick and dirty implementation on http://waddenzee.duckdns.org/, the icons on the left.

[image]

What do you think of this?

wellenvogel, 2024-02-08:
For sure an interesting idea.
One of the major issues is always to find a nice usable workflow - especially on touch devices.
And currently the number of buttons at the nav page is already close to the limit.
So potentially it would require some new concepts there.
In the moment my major focus is at the new ocharts implementation - maybe afterwards...

wellenvogel, 2024-02-08:
>How are the charts actually rendered? Do you use an opencpn lib for that? If yes it would also be capable of renderering unencrypted ENCs like available from NOAA.

Currently the o-charts_pi plugin is used. Are you sure that it could render unencrypted ENCs (i.e. did you ever do it with the plugin)?
I don't think so. OpenCPN itself can do this.

But any way the new ocharts plugin for AvNav will just reuse the license management and some basic stuff from the OpenCPN plugin - and it will be able to render S57 charts - after a conversion to the senc format.

quantenschaum, 2024-02-08:
No, I don't know much about these details. The ocharts just look like in opencpn, so I thought the renderer of opencpn is used and since opencpn can display unencrypted S57 ENCs avnav could too somehow.

quantenschaum, 2024-02-09:
The way you made it touch-operable, the big buttons (re-sizable!) and selection of points by clicking/touching or moving the map center to the desired location is superb! That really works in shaky conditions. Especially in rough and wet conditions I found plotters with real buttons like the B&G Zeus better to operate, touch alone might be difficult then but touch is faster and more comfortable in easier weather. My personal experience.

So, what about an additional set of buttons for other actions appearing near the touched location when you touch/click and hold a point on the map (not moving or zooming, just hold for short time)?

These buttons are of course hidden and you have to RTFM or try it to find out that it works like this. It's like clicking on the RTE widget in the left to edit the route. I would not have thought of it working this way, but once you know, it's perfect.

nmeyne, 2024-08-06:
> What do you think of this?

A nice set of features, but maybe implement them selectively?    Personally  I think the focus needs to be kept on graphical navigation on the tablet, without clutter, assuming a backup GPS in case of primary GPS fail.   If the screen / tablet fails than a PAPER chart is our back up for plotting rather than trying to replicate all the paper chartwork on a 'dumb' screen.

So, of these nice features, rather than have fixes / EP's, I think I would like to focus on functions useful in 'graphical' screen navigation that are missing from AvNav:  perhaps a course to steer line, from current position, by adding a tide/leeway vector.

For 'backup' navigation we still transfer GPS positions to our paper chart in pencil every 30mins or so, mark the time and course on it in and make an entry in the log.  

Sometimes bearing and distance off enhancement is good as a fast / easy way to transfer a position fix from AvNav to the 'backup' paper chart:

1. Find a reference point on the chart with a compass rose printed on it....  or a 'foolproof' point to place a breton plotter or similar
2. Put a special waypoint / mark  in AvNav corresponding to this point
3. Put the  display center cross on the special mark / waypoint so that AvNav continuously gives bearing (T) and distance off this reference point
4. Use a strip of paper calibrated (from the latitude scale) with NM to set on the compass rose / reference point on the chart to quickly mark the position on the chart.  Write in the time and COG. Lat and long in the log.

quantenschaum, 2024-08-25:
>A nice set of features, but maybe implement them selectively? Personally I think the focus needs to be kept on graphical navigation on the tablet, without clutter

I could be configurable when and how to show these tools or just have them as an optional plugin.

>assuming a backup GPS in case of primary GPS fail. If the screen / tablet fails than a PAPER chart is our back up for plotting rather than trying to replicate all the paper chartwork on a 'dumb' screen.

The whole thing is also interesting from the educational point of view and it's fun to build it.

And in the very eastern parts of the Baltic Sea GPS may not work currently or gives wrong positions. The GPS unit may be fine, but the signal is bad. 

So, why not turning off the GPS in that case an do the nav work manually as on paper. You can of course use paper directly, but why not doing it digitally?

I would also be nice to have a dead reckoning mode in the chart plotter where it start with a given fix an projects it using on board sensor data (compass, log, wind, clock).

>perhaps a course to steer line, from current position, by adding a tide/leeway vector.

That is what you could do with these tools. - If you have a working GPS, just make your COG match the bearing to the next WP and read you compass.

>Sometimes bearing and distance off enhancement is good as a fast / easy way to transfer a position fix from AvNav to the 'backup' paper chart

This can already be done using the measurement tool.

quantenschaum, 2024-08-25:
Things I would like to have which may be relatively easy to implement.

1. show bearing and reverse bearing in the measurement tool (for bearing from boat vs bearing from object)
2. (allow to) show magnetic bearing value to be able to directly draw a compass bearing line
3. allow multiple measurement lines be on the chart simultaneously, allows making a cross bearing