wellenvogel/avnav #323: Feature Request: AIS relative motion vectors

quantenschaum, 2024-02-12:
It would be nice to have the possibility to display relative motion vectors for AIS targets. The transformation is trivial and the usefulness for collision avoidance is great. All commercial radar/AIS systems support that.

There could be a shortcut for switching between true/relative. Or one could enable relative vectors additionally in the AIS settings and show them as dashed lines. Additionally there could be a marker on the relative motion vector at the location of the CPA.

wellenvogel, 2024-02-13:
What exactly do you mean with "relative motion"?

quantenschaum, 2024-02-13:
The motion vector of the other vessels (ais targets) relative to own ship. 

`relative motion vector = true motion vector - true motion vector of own ship`

These are the vectors usually shown by a radar with [ARPA](https://en.wikipedia.org/wiki/Automatic_radar_plotting_aid) that tracks contacts or as you would draw them on a maneuvering board. Using them you immediately see when on collision course: the relative motion vector is pointing at you. And they tell you, how to change your course and speed to open up the CPA.

more on this topic and theory can be found here (page 59) https://msi.nga.mil/Publications/RNMB

[image]

There is no need for a separate view looking like a maneuvering board, the relative motion vectors could be displayed on the chart directly but visually distinguishable for true motion vectors (like dashed). You just have to be aware of how to read these vectors, they do *not* a track on the map. Switching to heap up mode is useful here as well. Maybe they are disabled by default to avoid confusion but can be activated by the user if they want to.

It could look something like this.

[image]

quantenschaum, 2024-02-13:
If you could point me to the relevant sections in the code, I would try to implement it myself.

wellenvogel, 2024-02-14:
For the computations:
viewer/nav/aisdata.js
viewer/nav/navcompute.js
For the display:
viewer/map/aislayer.js
For the switch:
viewer/util/keys.jsx
viewer/gui/settingspage.jsx
Potentially some more for the ais list and info page.
Important to think about the error scenarios...

quantenschaum, 2024-02-15:
Is there a way to load the sources directly into the browser without processing the through the gradle whatever thing? How can I change the code and immediately run and debug it in the browser?

quantenschaum, 2024-02-15:
`../gradlew debug -t` in `viewer/`?

wellenvogel, 2024-02-16:
No.
At least you need to run the webpack build using
 `npm run warch`
In the viever dir.
In parallel start the server from within the forked git ( cd into the base dir):
 `linux/avnav -a . -v viewer/build/debug`
After any code change the webpack compiler will run and you can reload the page.
And you have the debug build that you can easily debug with the browser developer tools.

quantenschaum, 2024-02-16:
Thanks. OK, will try that.

Why so you draw dashed by drawing every single dash and do not use `context.setLineDash()`?

wellenvogel, 2024-02-16:
In the beginning I was trying to remain independent of the context drawing API to be prepared for different back ends (like WebGL or something OpenLayers specific). Maybe this could be changed as I don't think I will still go for any other drawing back end.
Finally some of the functions simply need some trial to get an impression what provides best visibility.  We always have to consider poor visibility due to heavy sun light or other factors.
So it's always a good idea to really try out things on a mobile device being outdoor.

quantenschaum, 2024-02-16:
So, here is my take on it. Have a look, please. 

It's a really cool project and nice to work and play with. Why didn't I find it earlier? Actually I had the idea of writing something like AvNav as well. It started with a server/proxy for tile based charts for use in OsmAnd, then I thought of adding own ship's position and other data to the map, but didn't have the time. Then I found AvNav. Someone already did the job. Very nice. Thank you.

wellenvogel, 2024-02-16:
Thanks a lot.
Will need some time as currently I'm busy with the o-charts stuff.

quantenschaum, 2024-02-16:
Thoughts on AIS data, updating, CPA computation.

- CPA should be (re)computed whenever the own ship's data is updated or ais target's data is updated since both enter the equation, so at every position updated (`updatetimes.position`)
  - CPA *must* be computed with COG/SOG of own ship and target always, never heading or waterspeed (can be misleading and dangerous)
- the dead reckoning position ("ghost") of the AIS targets should be updated at this pace, too (`updatetimes.position`), more often is ok
- AIS target's data gets updated after `updatetimes.ais` interval
- using HDG for COG is a very bad idea IMHO, imagine a vessel fishing or with restricted maneuverability, drifting sideways HDG!=COG (have seen that quite often), the course vector would point into a completely different direction, CPA would be totally wrong, this is dangerous. It is ok to use COG for HDG if no HDG is available, but never vice-versa. POS and COG+SOG is the minimum information the must be in the AIS data. I would remove that option.

example

QM2 HDG=010 STW=0 SET=280 DFT=3 COG=280 SOG=3 
both images show the *same* situation: left using COG, right using HDG

[image]

- CPA seems to be computed using COG/SOG always, this is good
- motion vectors using HDG are simply wrong (computed correctly using HDG, but the result is useless/dangerous for navigation)
- orientation of AIS target icon should always be based on HDG, fallback to COG possible

quantenschaum, 2024-02-16:
drifting vessel with HDG!=COG displayed correctly

[image]

wellenvogel, 2024-02-16:
For the HDG part:
Refer to #272

quantenschaum, 2024-02-16:
HDG stuff - What Aldebaran says is exactly what I said. The problem in AvNav is, that both the target icon and the course vector use either HDG or COG (as selected in the settings) simultaneously. They always point into the same direction. This is wrong!

- the target icon _should be oriented by HDG_ or COG if HDG is not available (as fallback, should be marked visually)
- the course vector (true motion vector) _must be oriented by COG always_, never by HDG :exclamation: 
 
The true motion vector is the _estimated track over ground_ for the selected time span, it is not related to HDG at all. Using HDG for its direction as in the example above yields completely meaningless line that suggests a movement of the target that actually does not exist. This is why this is dangerous. It does not make sense to have a `useHDGasCOG` setting.

Please think about it and look at the images I posted carefully.

wellenvogel, 2024-02-16:
Will just have a look...
Anyway if I remember correctly (I had a lot of discussions with him offline) - one of the issues is that COG can be completely wrong if your speed is very low (close to 0) as it is computed from GPS position differences. So at least this also has to be considered for the course vector. And what he says in your excerpt is slightly different - he does not like the option to let the symbol point to COG - it's not about the course vector.
Remaining problem:
Breaking changes is something I always try to avoid as much as possible. So at least we need to consider the potential consequences.

quantenschaum, 2024-02-16:
>one of the issues is that COG can be completely wrong if your speed is very low

Yes, but this is no problem because you always have to see (COG,SOG) as vector. A zero length (or very short) vector has effectively no orientation, the value of COG with SOG=0 does not matter. This nonsense COG is an artifact of the polar coordinates. The true motion vector displayed will be very short in this case, so the direction does not matter.

>he does not like the option to let the symbol point to COG 

Yes, because it doesn't make sense, but COG as fallback would be ok.

>it's not about the course vector

It is, but indirectly. If the tug he describes goes astern, having the true motion vector point ahead is wrong. Ask him about it, I know what his answer will be.

>Breaking changes is something I always try to avoid as much as possible.

Can understand that, but what AvNav displays as true motion vector when using HDG for COG is rubbish and dangerous. Consider the consequences of this. I would not hesitate to fix this.

You want to make AvNav as useful as possible with the data that is available. Superb! But it doesn't make sense do display data, just to display something even if it is wrong. _IMHO it's better display nothing at all, than to display wrong data._

Computing true motion vector from HDG is only correct if

- ship is going ahead
- does not make leeway or drifts
- there is no current/tide, water does not move

These assumptions are almost never satisfied, especially not

- in tidal waters
- due to leeway
- fishing or working vessels drifting sideways
- tugs with Voith-Schneider drives that can move in any direction

I cannot think of any convincing argument for computing the true motion vector from HDG, it's just wrong. If you don't have a (COG,SOG) vector, you cannot display a true motion vector, end of story.

It's a bit like the problem with the [SailInstrument](https://github.com/kdschmidt1/Sail_Instrument/issues/9). Here the calculations were also wrong, COG/SOG/HDG/STW were mixed. If you don't have HDG/STW one can fallback to COG/SOG to computer ground wind instead of true wind, assuming COG=HDG. That is ok, an first of all it is less dangerous because wind speed is not used for collision avoidance.

quantenschaum, 2024-02-17:
It's a pretty interesting experience to see how different people implement these calculations, but noone gets it completely right. There's also a plugin for SK that computes derived data, it also contains errors. I would really like to test products from B&G, Raymarine, Garmin..., feed them with test data and see what they show. Here the power of open source comes into play and many people can check and contribute to the code. This is great.

The root cause of these errors seems to be the common misconception that a ship moves over ground into the direction of it's heading. This is generally not the case. There is current and leeway. Everyone holding a yacht master or the like knows how to work out a course to steer, usually graphically on paper. You calculate the (true) heading you have to hold to reach your destination at a certain bearing. This heading is usually not equal to the bearing (not talking magnetic variation here). So, it should be obvious that a ship doesn't move over ground into the direction of it's heading, but still it is commonly assumed.

quantenschaum, 2024-02-17:
Here an example of AvNav (with TMV from HDG) vs OpenCPN (has no such option), both fed the same data.

- CPA is computed correctly by AvNav
- but the displayed TMV does not match the situation - Why is there a collision warning if the ships just pass each other about 1nm apart? This is confusing. You clearly see, that the displayed TMV is wrong.

[image]

quantenschaum, 2024-02-17:
It even gets more confusing if you look at the DR position ("ghost" :ghost: ) that is displayed by AvNav (right) compared to OpenCPN (left).

[image]

quantenschaum, 2024-02-17:
I hope all of this in convincing enough to remove the use-HDG-as-COG-option.

- use HDG for orientation of the ship icon, if it is not available use COG, there is no need for the user select something
- use COG and SOG for the true motion vector, if they are not available then do not display a TMV

wellenvogel, 2024-02-17:
I just checked again in the code. Maybe I missed some points...
For AIS targets the course vector is always computed from COG. For the Icon rotation (i.e. symbol direction you can select HDG or COG, HDG falls back to COG if not available).
For the own ship we have the course vector (solid line) - always computed from COG. And we have the ship direction vector (dotted line, optional) that is based on the ship direction. The intention for this one is to easier align the chart view with what you see in reality if you look right ahead on the ship.
CPA computation is always based on GPS data (i.e. SOG/COG).
So what do you think is wrong?

Would be helpful to have some testing data available that shows a situation where AvNav does not compute it like this.

wellenvogel, 2024-02-17:
Additional remark:
I compared your PR code (aislayer) with the original code.
Except for the relative motion vectors you are now back to the original handling - only you cannot switch back to use COG for the ais target direction (what is there to maintain compatibility...).

quantenschaum, 2024-02-17:
>For AIS targets the course vector is always computed from COG. For the Icon rotation (i.e. symbol direction you can select HDG or COG, HDG falls back to COG if not available).

Does this describe how it currently is or how it should be?

I observed (examples above) and found in the code that the course vector (aka true motion vector) is computed using HDG if this is selected in the settings.

quantenschaum, 2024-02-17:
Maybe I really mixed your and my changed code. Sorry for the noise then, but then this is clarified fir sure now.

quantenschaum, 2024-02-17:
I readded the HDG/COG setting but added a fallback to COG if there is no HDG.

Sorry for the confusion, but now this is topic has been discussed thoroughly. So, if someone asks...

wellenvogel, 2024-02-18:
I made some tests with your pr code.
For me the view now looks rather confusing...
[image]

What I see:
(1) rot indicator is very dominant (and cannot be styled). And I do not really understand the formula for computing it.
(2) For me all the relative motion vectors (really far away from my current position) are at best confusing. Maybe it would make sense to limit them to selected/nearest/warning target - or to targets in a close range to me.
(3) for the dashed lines yellow is maybe not the best initial color (you can adapt this of course...)

wellenvogel, 2024-02-20:
Sorry I was not clear. The concept of the relative motion vectors is clear. And I like it.
But the problem (for me) was just with all the targets being far away from my own pos.
So I guess the solution with a range would really address this.

For the rot I'm not really sure if we should like to have it like this. Maybe a small indicator close to the ship symbol would be better.
The problem will all AIS stuff is the to find a balance between displaying a lot of information and making the chart unreadable if we have many AIS targets.
I know about the value in the spec.But I could not really get the point for:
`rot/300*target_sog*courseVectorTime`
as length of the vector.

quantenschaum, 2024-02-21:
`rot/300*target_sog*courseVectorTime` = length of TMV scaled by rate of turn, 300 is just for normalization, was just a try to make the black longer when the ship is turning faster

quantenschaum, 2024-02-22:
I added optionally curved TMVs and RMVs. The estimated true and relative track is drawn including the rate of turn information and shows up as a curved line if the target is turning. This is very valuable information in situations when a big ship is turning slowly. This is currently not possible to see in avnav since there is neither a turn indicator nor is the rate of turn displayed in the AIS data. You would have to observe how a target changes its course between AIS updates. The curved paths can show you potentially dangerous operations right away long before a CPA warning is displayed (CPA computation doesn't consider ROT).

example
- :arrow_right:  straight track w/o ROT, :arrow_left:  curved track including ROT
- the upcoming collision is immediately visible, RMV hits own ship
- you see right away that and how the target in turning
- the CPA warning is issued very late.

[image]

https://github.com/wellenvogel/avnav/assets/3985599/7450167e-b7a4-4044-a82c-b1e723473390

quantenschaum, 2024-04-09:
is implemented

closed 2024-04-09 by quantenschaum