wellenvogel/avnav #327: Map Zooming

oxbown, 2024-02-20:
Is there any Map setting in AvMap to force the zoom of maps to int values and not decimal values? Is useless for tiles maps.

wellenvogel, 2024-02-20:
No it's not.
And currently at the browser side the maps are always raster.
Not sure how much you already used it - I found it convenient to be able to slightly increase symbols/text with the intermediate zoom levels some times .
But it could be a nice improvement to lock this by a setting.

quantenschaum, 2024-02-21:
>I found it convenient to be able to slightly increase symbols/text with the intermediate zoom levels some times

overzooming is helpful indeed (to read a small label), but locking to integer zoom levels (native resolution) makes the map look nice and crisp 

I actually had different question about zooming the map: Is it possible to add single finger zooming :question: 

It works in many applications (google maps) and is very useful when using+holding the device with one hand (thumb only). You double tap and hold  (down-up-down+hold), then you slide the finger up/down to zoom.

wellenvogel, 2024-04-23:
I now went for a slightly changed approach:
You can force the map to be on int zoom levels when you use the zoom buttons (on by default).
Maybe this is a working compromise as I guess limiting the 2 finger zoom to int levels will feel rather ugly compared to the current behavior.
And btw: typically scaling up the chart to a small degree (< 2) still gives reasonable results. Therefore you also have the map scale option for raster charts as sometimes the texts are simply too small using the normal resolution.

closed 2024-05-20 by wellenvogel