wellenvogel/avnav #328: ochart rendering - cut off labels and symbols

quantenschaum, 2024-02-22:
First of all it's really cool that ocharts can be used in avnav. It works really well and stabe so far. Thanks a lot for this feature!

What I observed is that some objects like labels and light sectors are cut off. I think you probably know that already and that it has to do with the tile rendering. If a vertex is outside the currently rendered tile, the tile does not contain any data of this vertex even though there would and should be rendered something from this vertex on this tile. This a common problem when rendering tiles, which can be addressed with [meta tiling](https://mapproxy.github.io/mapproxy/labeling.html#wms-labeling). MapProxy has the options [`meta_size` and `meta_buffer`](https://mapproxy.github.io/mapproxy/configuration.html#globals) to tune this when generating tiles from a WMS. 

This could be done in avnav too to mitigate this kind of problem, couldn't it?

- light sectors are often cut off and 
- are far too small to be useful in general
- light characteristics are sometimes missing or incomplete, the color of a single sector is stated only, seems like it tries to render a label for each for each sector but one of them displayed in the end

some examples comparing ocharts to ["my" map](http://waddenzee.duckdns.org/#8/54.265/9.198) (rendered with meta tiling and buffer enabled)

cut off light sectors, only color W in the label, but it actually is RGW    
[image]

cut off label 
[image]

cut off sectors, missing light characteristics  
[image]

cut off sectors, incomplete light characteristics (there's a fixed sector _and_ a flashing 360 light)
[image]

lightsectors too small  
[image]

free-x, 2024-02-22:
With "ochartsng" it looks much better ;-)
Did you tried?
[image]
[image]
[image]

Although I would admit that the light sectors are too short

quantenschaum, 2024-02-22:
Very nice. Is this is still work in progress? How can I get this running?

- The sectors are still too short, yes.
- The labels still show only one sector as it seems.
- It seems not to get clipped anymore.

Just out of curiosity: How does the rendering work? 

- charts are encrypted in the oesu files
- lib from ocharts derives the decryption by some secret magic from hardware and os properties
- ocharts lib provides an api to access the decrypted vector data
- some lib (which?) renders this vector data on the screen or into a buffer/file
- the screen/buffer is captured/converted and cut into XYZ tiles and served to the browser

Does is work like that? Which lib does the actual rendering? And to what extent are you actually able to influence the appearance of the rendered chart?

free-x, 2024-02-22:
> Very nice. Is this is still work in progress? How can I get this running?

Sourcecode you can find here on github
https://github.com/wellenvogel/ochartsng

Since last Saturday is public beta available
https://www.segeln-forum.de/thread/90260-avnav-ocharts-jetzt-auch-f%C3%BCr-android/
https://www.wellenvogel.net/software/avnav/docs/hints/en_ochartsng.html?lang=en
Are you stil OpenPlotter 3 user? Right?
If yes, you need to activate preview repository
```
deb [signed-by=/usr/share/keyrings/oss.boating.gpg] https://www.free-x.de/deb4op bullseye-preview main
```
Then
```
sudo apt update && sudo apt install avnav-ochartsng
```

free-x, 2024-02-22:
BTW...ochartsng can also display S57( ok, after conversion [to SENC](https://github.com/wellenvogel/ochartsng/blob/master/tools/tosenc.py))

Here is Screenshot of Waddenzee from RWS
[image]

quantenschaum, 2024-02-22:
Thank, just looking into it. Impressive!

wellenvogel, 2024-02-22:
> Just out of curiosity: How does the rendering work?
charts are encrypted in the oesu files
binary lib from ocharts derives the decryption key by some secret magic from hardware and os properties
ocharts lib provides an api to access the decrypted vector data
some lib (which?) renders this vector data on the screen or into a buffer/file
the screen/buffer is captured/converted and cut into XYZ tiles and served to the browser
Does it work like that? Which lib does the actual rendering? And to what extent are you actually able to influence the appearance of the rendered chart?

Depends on the version...
The "old" ocharts support completely used the OpenCPN o-charts_pi plugin.
The new one uses an own implementation (with some code taken over from the old one and from o-charts_pi).
The old one was using the X11 infrastructure for rendering (including OpenGL).
The new one does all the rendering completely in memory.
And both are just rendering single tiles. Many of the problems with the old implementation are caused by this fact.

For customizing:
You can adapt a lot of things via the settings.
And finally you can dramatically change a lot with editing the xml files. I just plan to make this a bit more user friendly (i.e. having a separate user.xml to be layered on top of  the existing settings).

quantenschaum, 2024-02-22:
Awesome work!

Just quickly one thing I noticed: Spot soundings are missing on drying heights.

new -- old  
[image]

wellenvogel, 2024-02-22:
OK - I can also see this issue.
Thanks for finding this!

wellenvogel, 2024-02-22:
New daily tonight.

quantenschaum, 2024-02-22:
display category = all

another thing: too many digits after decimal for topographic heights or mag variation

[image]

[image]

quantenschaum, 2024-02-22:
floodlight icons are out of place

[image]

quantenschaum, 2024-02-23:
I will create new issues there. - Don't see it as complaints, just feedback, so you know about it.

Sure, I would test it on android, if that helps you. I just installed the `avocharts-beta-20240222` and have `avnav-20230705` on android 13. I tried to add s57 charts (zip with senc files from opencpn), but when clicking on "upload zip" nothing happens, no file chooser opens.

wellenvogel, 2024-02-23:
for the first 2 points I already created issues.
If you launch the UI from within AvNav you don't have a file chooser.
Just launch it directly from the avocharts app - or connect to AvNav with a browser (e.g. from your PC - easier upload any way).
I will extend AvNav itself later for a file browser.

quantenschaum, 2024-02-23:
that worked

quantenschaum, 2024-03-25:
Moved to ochartsng repo

closed 2024-03-25 by quantenschaum