wellenvogel/avnav #335: How to change the ipadres of wlan0

Fietsenwiel, 2024-04-10:
I installed an new image of AvNav.
The ip adres of wlan0 is 192.168.60.10.

I like to change that and I see you can change that in: 
/etc/network/interfaces.d/wlan0

Its working (ifconfig shows the new ipadres) but connecting is not possible anymore.
So I think it has to be change anywhere else too, maybe the dhcp server?

Can you please help me in the right direction?

free-x, 2024-04-10:
don't know why you need this to do

- /etc/network/interfaces.d/wlan0
- /etc/dnsmasq.d/avnav-dnsmasq.conf
- /etc/iptables.ipv4.nat

Don't forget to reboot

Regards
free-x

Fietsenwiel, 2024-04-10:
Thanks for your fast reaction.

I changed 192.168.60.10 in my own ipadress (10.10.10.1)
In /etc/dnsmasq.d/avnav-dnsmasq.conf of course only 192.168.60 for 10.10.10
But then still I can't connect...

Thanks in advance.

Greeting

Lschip

free-x, 2024-04-10:
Could you provide Output of "ifconfig", "route -n", your changed config files?

Regards
free-x

wellenvogel, 2024-04-10:
Could you explain what you would like to achieve?
There is alot of players in the game that need to function correctly.
How do you connect when making the changes?

Fietsenwiel, 2024-04-10:
Hi Andreas,

I always used this host adres (10.10.10.1) for connecting with the
tablet(s) to the raspberry pi with AvNav.
It's in my documentation, qr code on the boat and I have an arduino with
this host adres fixed programmed.

So I thought it is easy to change the host adres and everything else stays
the same...

Thanks in advance,

Lschip

free-x, 2024-04-10:
Do I understand correctly that you are migrating from OpenPlotter to AvNav Image?

wellenvogel, 2024-04-10:
Sorry l don't get this completely...
How did you set up a pi in the past to use this address?
Finally you have to change the interface addresses, the dhcp config, the iptables stuff - as free-x mentioned.
It's much simpler to rely on Bonjour aka avahi to connect.
This way you can even change the ip setup without affecting the clients.

Fietsenwiel, 2024-04-10:
> @wellenvogel <https://github.com/wellenvogel> means your
> avnav-dnsmasq.conf should look like this
> interface=wlan0,wlan-ap,wlan-ap2,wlan1
> domain=avnav
> dhcp-range=wlan-ap,192.168.60.20,192.168.60.254,255.255.255.0,12h
> dhcp-range=wlan0,10.10.10.20,10.10.10.254,255.255.255.0,12h
> dhcp-range=wlan-ap2,192.168.40.20,192.168.40.254,255.255.255.0,12h
> dhcp-range=wlan1,192.168.50.20,192.168.50.254,255.255.255.0,12h
> dhcp-option=42,0.0.0.0
> addn-hosts=/usr/lib/avnav/raspberry/hosts

Ok, I changed that according to your latest mail.
I can see the accespoint but when I try to connect it does not. Still the
same problem....

Fietsenwiel, 2024-04-11:
> Could you please provide the info I have asked for - i.e. the client set
> up when connected to the AvNav wifi?
> Ip, routes...

I'm lost now.
I restored the original files but the problems stays.
Now even the ap disappears and comes back al the time, connecting is not
possible.

i now see this:
***@***.***:~ $ ifconfig
wlan-ap: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.30.10  netmask 255.255.255.0  broadcast 0.0.0.0
        inet6 fe80::20f:55ff:fea8:9e0  prefixlen 64  scopeid 0x20<link>
        ether 00:0f:55:a8:09:e0  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 243  bytes 70167 (68.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

***@***.***:~ $ route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use
Iface
default         RT-AC87U-19B0   0.0.0.0         UG    1      0        0 eth0
default         RT-AC87U-19B0   0.0.0.0         UG    10     0        0
wlan-av1
link-local      0.0.0.0         255.255.0.0     U     1000   0        0 eth0
192.168.0.0     0.0.0.0         255.255.255.0   U     0      0        0 eth0
192.168.0.0     0.0.0.0         255.255.255.0   U     0      0        0
wlan-av1
192.168.30.0    0.0.0.0         255.255.255.0   U     0      0        0
wlan-ap

It looks good.

The ap shows as "relexed1" while before it was "relexed"

I dont want to consume too much time off you so else I install from scratch
(but is a lot of work with extra installed functionality)

Thanks in advance!

wellenvogel, 2024-04-11:
For me it's still not really clear what you are trying to do.
What do you mean with:
`the ap disappears and comes back al the time, connecting`?

Do you talk about connecting to the AvNav wifi?
Or about connecting with a browser?
Would be really helpful to get a clear description about what exactly you are doing and what exactly does not work.

Fietsenwiel, 2024-04-12:
Ok, I am sorry not to be clear enough.

First of all, it works again with the original configuration. First it
didn't but my wifi dongle was in the wrong usb port, I corrected that.

I want to connect with a tablet to the Raspberry py to get AvNav on it. And
also with a nodemcu (esp8266).
Now, the AvNav wifi accespoint has an ip adress of 192.168.60.10. And I can
connect with the tablet on it. So no problem at all anymore.

But the only thing is I want is change that hostadres (192.168.60.10) to
10.10.10.1

The most important reason is that I have this ipadress (10.10.10.1) hard
coded in a nodemcu and it is hard to change that. I thought it is easy to
change the ip adres on the Raspberry pi. But that looks to be wrong....

free-x, 2024-04-12:
does nodemcu get 10.x.x.x network assigned via DHCP? or is this also hardcoded?

Fietsenwiel, 2024-04-12:
It is hard coded...

wellenvogel, 2024-04-12:
Ok, first part understood.
But the second part is still open:
Would be really helpfull to get a clear picture what you did and what EXACTLY did not work as exoected.
"Does not connect" is not really a clear description....
So after you changed something it would always be good idea to try with a client where you have some analysis capabilities and check step by step:
Wifi connect, IP address, routing, ping, browser
And document the results.
As said: it's more then just the ip.
And again: using dhcp and bonjour is extremly simple even on a node mcu...

free-x, 2024-04-12:
@Fietsenwiel 
what happens when you create an alias on wlan-ap?
```
sudo ifconfig wlan-ap:0 10.10.10.1 netmask 255.255.255.0 up
```

Fietsenwiel, 2024-04-12:
This doesn't work, I don't have a wlan-ap.

But I am going to start from scratch by installing Bullseye, configuring
the wlan like I need and then install avnav on it.
I think this is the best way for me to go, then I know how the
configuration works.

Thanks for your help anyway.

wellenvogel, 2024-04-12:
So most probably in this case you  do not want to install avnav-raspi - as this would set up the Ip as in the image.

Fietsenwiel, 2024-04-12:
Yes, maybe that is the richt conclusies.

I want to make more charges like a vpn to my router so that I can acces the
raspberry from home.

So then the avnav image is maybe not the best idea.

Thanks for your help!

free-x, 2024-04-12:
> i now see this:
***@***.***:~ $ ifconfig
wlan-ap: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.30.10  netmask 255.255.255.0  broadcast 0.0.0.0
        inet6 fe80::20f:55ff:fea8:9e0  prefixlen 64  scopeid 0x20<link>
        ether 00:0f:55:a8:09:e0  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 243  bytes 70167 (68.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

> This doesn't work, I don't have a wlan-ap.

what now? did you or didn't you? is very confusing, and is too late anyway after your decision

BTW: which RPI do you have? is not clear to me anyway, how is it possible to initialize "wlan0" with our image... Internal WiFi is always assigned "wlan-ap"

wellenvogel, 2026-11-02:
implemented with [20261102](https://www.wellenvogel.de/software/avnav/newDoc/latest/en/installation/versions/20261102.html)

closed 2026-11-02 by wellenvogel