wellenvogel/avnav #337: Feature request, Route numbering

AaskovKnudsen, 2024-04-26:
Hi Andreas
After creating a route, and adding additional waypoints, the numbering of waypoints,  might not be in a logical line.
It would be nice with a botton to rearrange the numbering from 1 to x. Might also work after inverting a route.
Thank you for creating avNav.
Regards Jan

AaskovKnudsen, 2024-04-27:
I would suggest to rename from 1 to n nomatter what.
If you want to keep the wp name, don't push the botton.
Regards Jan

antonical, 2024-07-21:
Can I send that. For instance I created a route from Grimsby to Hartlepool but then added stops into Scarborough and Whitby, That ended up with lot of new points out of sequence. It would be great to have a button to rename them back to linear order. 

Follow on question If I have named a number of waypoints with custom names is it possible to leave them but put them in sequence as the order hasn't changed. See screenshot as an example.

file:///home/tickers/Pictures/Screenshots/Screenshot%20from%202024-07-21%2012-23-58.png

wellenvogel, 2024-07-22:
Not sure if I understand correctly what your intention is with the waypoints with custom names.
Currently there is no way for AvNav to know (in a reliable way) if a point in the route has an automatic name - or if you provided a custom name.
This would require a (non standard) extension in the stored gpx.
So for now a complete renumbering will be straight forward. But keeping custom names is more complicated.

antonical, 2024-07-22:
HI Andreas,

Just renumbering them into a linear sequence would be great.

Cheers

wellenvogel, 2025-07-23:
release with [20250723](https://www.wellenvogel.de/software/avnav/downloads/release/20250723/?lang=en)