wellenvogel/avnav #339: Overlay markers not shown when using user icon

ukoehler, 2024-05-21:
Version: AVNav Version 20230705
Openplotter

I added a GPX as an Overlay. When choosing 

icon file: empty and default icon: empty I can see the markers
icon file: empty and default icon: DefaultGPXIcon I can see the markers
icon file: see attached and default icon: DefaultGPXIcon I cannot see the markers

The marker sym element is <sym>Activity-Diving-Snorkel</sym>
The png is stored in the Zip as Activity-Diving-Snorkel.png without a path

ukoehler, 2024-05-21:
[OpenCPNMarkerIcons.zip](https://github.com/wellenvogel/avnav/files/15387770/OpenCPNMarkerIcons.zip)
[Dive 2023.zip](https://github.com/wellenvogel/avnav/files/15387777/Dive.2023.zip)

I had to add the gpx in a ZIP file, since it was rejected otherwise

ukoehler, 2024-05-21:
It works when using the SVG or PNG as DefaultGPXIcon

wellenvogel, 2024-05-22:
I cannot confirm this issue.
Just tested it with the files you provided and all combinations work as expected.
Could you retest this with the current version (although no changes there...)?
For the last combination: the icons from the icon file will always win.

closed 2024-05-24 by ukoehler