wellenvogel/avnav #340: Request: support svg as user marker icons

ukoehler, 2024-05-21:
According to the documentation, I cannot use SVG files as user marker icons, but as default GPX icons.

[doc](https://www.wellenvogel.net/software/avnav/docs/hints/overlays.html)

It would be nice to be able to use SVGs as user icons, as well.

I cannot try this due to #339

wellenvogel, 2024-05-25:
Would be good to know what exactly you did. What do you mean with "a restart of the service"?
For testing see my comment https://github.com/wellenvogel/avnav/issues/341#issuecomment-2131308519.
And please attach the used gpx and icon file.

ukoehler, 2024-05-25:
OK, I reverted the changes after the behaviour changed today (see comment #341). The files were the same as used there with two changes:

In the icon file all PNGs renamed.
Changed width="64" height="64" from 18mm each.

I still do not get how it displayed the icons after I deleted the icon file.

wellenvogel, 2024-05-25:
How do you add/delete the files?
Do you do this from with AvNav (i.e. on the download page)?

ukoehler, 2024-05-25:
Yes, I always go through the download page. However, I check the files in the folder directly for content or to see if it did get deleted.

wellenvogel, 2025-03-10:
Closing this due to no further activity.

closed 2025-03-10 by wellenvogel