wellenvogel/avnav #343: US ENC charts with AvNav Android

ibravo, 2024-06-02:
Writing these instructions for future users. It might be good to include them under the Android section of the documentation:

If you want to use US ENC charts, NOAA provides free MBTiles that contain raster datasets derived from the latest NOAA ENC data. To download MBTiles for a region, click the region in the map, click on the MBTiles link, and save the .mbtiles file to your compute. The url to download the files is located at https://distribution.charts.noaa.gov/ncds/index.html

Once you complete the download, use Files to move the download to your user-defined charts location. Then click on the UP arrow to upload to AvNav Android application. Once finished, you will the mas available as a new overlay.

wellenvogel, 2024-06-05:
Thanks for this hint. Will include some info in the doc.
Btw: With the new [ochartsng plugin](https://www.wellenvogel.net/software/avnav/docs/hints/ochartsng.html?lang=en) you can also use the S57 charts...

wellenvogel, 2024-06-16:
added some hints in the documentation: https://www.wellenvogel.net/software/avnav/docs/charts.html?lang=en#h2:ChartSources

closed 2024-06-16 by wellenvogel