wellenvogel/avnav #344: WebP support for charts

quantenschaum, 2024-06-11:
I just tried the following

1. take an `mbtiles` chart file
1. [convert](https://github.com/quantenschaum/mapping/blob/master/convert.py) the binary blobs containing the image data to lossless [WebP](https://en.wikipedia.org/wiki/WebP)
1. drop this file into the `charts` folder of avnav
1. use the chart

:tada: And it works out of the box!

The webserver serves the files with `png` extension and as `Content-Type: image/png` but it still works (at least with Chrome and Firefox).

Using WebP results in a massive reduction of the file size of the chart files without any loss in quality (490MB -> 92MB, 19%).

It even kept working after converting the `mbtiles` to `gemf` (it's just binary blobs, the decoding happens in the browser).

The creation of the `mbtiles` files its actually not in the scope of avnav itself, but nonetheless I think this is worth mentioning.
Unfortunately it does not seem to be possible to instruct mapproxy to write `mbtiles` containing WebP images.

wellenvogel, 2024-06-11:
That's cool.
For the mapproxy part:
Maybe we could extent the plugin handling that I already have. For some sources you need a bit of python code to work properly...

quantenschaum, 2024-06-11:
Since MapProxy is written in Python it might be possible. Encoding WebP requires more CPU power, but once it's in the cache that doesn't matter any longer. Decoding happens on the client anyways. 

WebP GEMF files do even work on the android app (at least on my phone).

Another thing with MapProxy is, that it creates non-standard mbtiles with the y-axis inverted.
According to the [spec](https://github.com/mapbox/mbtiles-spec/blob/master/1.3/spec.md) mbtiles use the [TMS scheme](https://wiki.osgeo.org/wiki/Tile_Map_Service_Specification).  
This is always confusing. (for the record: TMS: y-axis up, sw origin, XYZ: y-axis down, nw origin)

WebP is a supported format im mbtiles according to the spec. OpenCPN does not seem to like it.

To be able to use mbtiles created by MP with other software, I added the `-y` option to my [converter](https://github.com/quantenschaum/mapping/blob/master/convert.py).

Trying to run your [`convert_mbtiles`](https://github.com/wellenvogel/avnav/blob/master/chartconvert/convert_mbtiles.py) directly on mbtiles created by MP does not work (neither `tms` nor `xyz` mode)

free-x, 2024-06-11:
for reading of xyz mbtiles you can choose corresponding option
[image]

quantenschaum, 2024-06-11:
OK, so that basically inverts the y axis. I used the command line script and it failed.

quantenschaum, 2024-06-11:
It's in the web ui. So, this is no for downloading the mbtiles file, it is to set the tiling scheme, such that avnav knows how to read it.

Does it only set flag or does it actually convert the file?

free-x, 2024-06-11:
Only set the read option

quantenschaum, 2024-06-11:
That's good to know. But I was also interested in creating mbtiles to be used in other software. Then you need to actually flip y in the file.

free-x, 2024-06-11:
I have used for example https://github.com/mapbox/mbutil

mb-util xyz.mbtiles xyz
mb-util --scheme=tms xyz tms.mbtiles

quantenschaum, 2024-06-11:
Thanks, more tools. - The process is fairly simple, I cooked my own converter for now. 

The issue was converting mapproxy mbtiles -> gemf, that did not work.

To my converter I also added

- reencoding images to another format
- discarding fully transparent tiles (then tiles from lower zoom levels get displayed in avnav)
- discarding small tiles like 1x1px placeholders (")

Depending on the data source that reduces the file size quite a lot.

quantenschaum, 2024-06-13:
Extending MP for use in avnav which may also result in a PR for MP itself sounds good.

Interesting fact: If MP gets fed with WebP images from an upstream tile source it actually does store these WebP blobs in the mbtiles db and also serves them but with a wrong `png` extension and mime-type (the data actually is WebP). This works as long as the tiles are just passed through in MP's internal processing chain, converting from TMS->WMS, the images gets reencoded as png. So, the [image handling lib (PIL) used by MP](https://github.com/mapproxy/mapproxy/blob/master/mapproxy/compat/image.py) does support WebP. 

Setting

```yaml
services:
  wms:
    image_formats:
      - image/png
      - image/webp
```

causes MP to server WebP via WMS. This does not work for TMS, though.

quantenschaum, 2024-06-13:
Since version 2 of MapProxy there is the `image` option on the cache which allows to change the image format.

```yaml
caches:
  c_foo:
    sources: [ s_foo,s_foo ]
    grids: [ GLOBAL_WEBMERCATOR ]
    cache:
      type: mbtiles
    image:
      format: webp
```

This allows the use of the `webp` file extension in the tile url. But the image is *not* converted from png to webp (using a png source upstream).

The demo web ui displays image format `webp`, but the layer in the viewer still uses `png`, so no tiles are displayed.

You can force MP to reencode the tiles by setting multiple sources on the cache. The MP has to merge the tiles and write the resulting image to the cache. And then it actually writes webp tiles to the cache (`file` as well as `mbtiles`). You can add the same source twice (downloading the tile two times).

quantenschaum, 2024-06-13:
As workaround you can define a dummy source pointing to MP itself, but an invalid url always returning 404 and mapping this to a transparent tile.

```yaml
sources:
  dummy:
    type: tile
    grid: GLOBAL_WEBMERCATOR
    url: "http://localhost:8004/dummy"
    on_error:
      404:
        response: transparent
caches:
  c_foo:
    sources: [ dummy,s_foo ]
    grids: [ GLOBAL_WEBMERCATOR ]
    cache:
      type: mbtiles
    image:
      format: webp
```

oxbown, 2024-07-09:
@quantenschaum have you tried AVIF format? 
Pillow does not support directly the format but there is a [plugin](https://github.com/fdintino/pillow-avif-plugin/) for that. The plugin uses the[ libavif library](https://github.com/AOMediaCodec/libavif).
In theory, AVIF files are 50% smaller than WebP and modern browsers support the format.

quantenschaum, 2024-07-09:
Interesting, will try this out. WebP is supported by mbtiles.

oxbown, 2024-07-09:
Just an example for size comparison:
```python
from PIL import Image
from pillow_avif import AvifImagePlugin
import os

local_dir = os.path.dirname(__file__)

with Image.open(local_dir + "/test.png") as im:
    im.save(local_dir + "/test.avif")
    im.save(local_dir + "/test.webp")

print("PNG Size:" + str(os.path.getsize(local_dir + "/test.png")) + "KB")
print("WEBP Size:" + str(os.path.getsize(local_dir + "/test.webp")) + "KB")
print("AVIF Size:" + str(os.path.getsize(local_dir + "/test.avif")) + "KB")
```
`PNG Size:4836KB
WEBP Size:3888KB
AVIF Size:2096KB`

quantenschaum, 2024-07-09:
Does AVIF have a lossless mode? That's what I prefer here because the charts should look clear and crisp w/o compression artifacts.

quantenschaum, 2024-07-09:
The AVIF encoded charts looks good. Encoding if very slow. I will transcode a 200MB WebP mbtiles file and then check the file size and compatibility with with different apps. In the browser it works fine.

oxbown, 2024-07-09:
Lossless attribute seems not work, but the codec parameter "svt" makes even more efficient results:
```python
im.save(local_dir + "/test.avif",codec='svt')
````
`PNG Size:9925KB
WEBP Size:3844KB
AVIF Size:10854KB`

oxbown, 2024-07-09:
Seems that Avif dont have a lossless mode. I made a few tests and the results are quite interesting. Sometimes AVIF wins over WEBP and in others not... 
This makes me a question.... can be made a mixed format chart?

quantenschaum, 2024-07-09:
>can be made a mixed format chart?

In principle yes, a browser treats each tile as a single image, other apps may also do this. In mbtiles you can only specify one single format, but if this field is not used and the image format is detected from the binary blobs themselves, it might work.

For simplicity I would stick to a single format, though. And you have to encode each tile twice to see which is smaller. And how much do you gain by doing this?

quantenschaum, 2024-07-09:
## Compatibility
- MapProxy: setting an upstream tile source that returns AVIF tiles makes MP serve AVIF tiles. Transcoding and converting to WMS does not work. 
- AvNav: Tiles are shown by the browser, if the browser supports AVIF, it will work with AvNav (also via MP)
- OsmAnd: Works via AVIF raster tile sources, but transparent areas are displayed as black
- JOSM: no, but does not work with WebP either
- OpenCPN: via mbtiles, no, but does not work with WebP either

quantenschaum, 2024-07-09:
Having 66% of the mbtiles files converted to AVIF and already exceeded the size of the WebP encoded file, I think it compresses better on some tiles, but not in general. But lest wait for i to finish and try again with `svt` codec.

quantenschaum, 2026-11-12:
WebP is effectively supported by the frontend.

closed 2026-11-12 by quantenschaum