wellenvogel/avnav #347: NMEA decoding, missing values treated as zero

quantenschaum, 2024-07-01:
When decoding NMEA sentences, sometimes values in a sentence are missing, no value is given, like `$GPVTG,,,,,,,,,N*30`.

AvNav [decodes this](https://github.com/wellenvogel/avnav/blob/bcb1e4322d6013a4defc4cf94f926be4490b3436/server/avnav_nmea.py#L335) to `{'track': 0.0, 'speed': 0.0}`.

IMHO this is wrong, because track and speed are not known, their values are undefined and not zero.

The fallbacks in the decoder like `float(darray[3]or '0')` should be removed. If the value is not present, then it must be omitted and not be mapped to zero. Undefined values should be set to `NaN` or `None` or should not be present in the dict at all. Then the values times out and gets removed from the internal storage model.

wellenvogel, 2024-07-01:
I agree. But this most probably requires some work as the consequences have to be checked carefully throughout the complete system.
As always there are at least 4 places:
(1) decoder (python)
(2) decoder (java)
(3) front end
(4) signalk handler

quantenschaum, 2024-07-01:
But if you do not write missing values to the data store, they time out and get removed, then it is like as have never been there in the first place. So, this should be not problem for the front end or SK.

quantenschaum, 2024-07-01:
side question: When looking into this I stumbles upon this;

COG, course over ground has the [key `track`](https://github.com/wellenvogel/avnav/blob/bcb1e4322d6013a4defc4cf94f926be4490b3436/server/avnav_nmea.py#L78) but in the GUI, in the widgets and int the data store it is available as `nav.gps.course`. This is pretty confusing. Why? 

SOG for instance is just `speed` everywhere.

quantenschaum, 2024-07-01:
Values in the data store should expire after a certain time if they are not updated. `nav.gps.course` does expire and gets removed from the store and becomes `undefined` in the in the frontend when access in widget code via `storeKeys: { COG: 'nav.gps.course' }`. Other values like `nav.gps.sail_instrument.COG` seem also to get removed from the store, but are still available in frontend code?

The COG widget displays `---` if there is no `course` present, but the course vector is still shown with direction 000.

wellenvogel, 2024-07-01:
yes, as I said - would need to go through all the places. 
Maybe even needs some ideas how to show course vector/boat direction...

quantenschaum, 2024-07-01:
Hmm, OK. - But a field becoming undefined (times out) can happen (sender goes offline or leaves the field blank).

Could you help me to understand why `nav.gps.course` becomes `undefined` in the frontend when remove from the store, but other fields like `nav.gps.sail_instrument.COG` do not?

quantenschaum, 2024-07-02:
Fields that have expired on the server are no longer present in the data returned in `/viewer/avnav_navi.php?request=gps` to the frontend. The field `course` is explicitly set to be `undefined`. When a widget requests data fields via the `storeKeys` dict, the expired fields are not removed from the widget's `props` and the old expired value is retained. 

Where does the updating of the widget's props happen in the code?

quantenschaum, 2024-07-02:
> some ideas how to show course vector/boat direction...

If COG/HDT is undefined
- do not display a course vector
- display boat icon as circle/dot (no orientation)

wellenvogel, 2024-07-02:
Most probably you are right - I have to go back to some older discussion about this stuff.
But we should not try to mix this with the undefined handling.

> Where does the updating of the widget's props happen in the code? I'd like to understand, how that works.

hoc/Dynamic.jsx
Very generic handling that updates properties with values from the global store if they are changing.
Needs some thoughts...
Maybe undefined is not really the way to go .
Currently I'm a bit busy with some other tasks (porting to reactjs 18 with a couple of pitfalls) and some stuff outside AvNav - so will still need some time until digging in deeper.

quantenschaum, 2024-07-02:
Sure, not in a hurry. - I came across this because I observed some unexpected values when my GPS was stationary (it omits COG when not moving) or when it lost its fix. COG was set to 0 in the NMEA parser. After changing this to not updating COG, it expires and is removed from the store. In `gpsdata.js` the `course` is explicitly set to `undefined`, so COG is displayed as `---` as expected. Interestingly this also works for other values as well which are not explicitly set to `undefined`, like HDT. After changing the default value of the HDT widget it also shows `---` when SOG expires. 

This is the behaviour I prefer. If the value is not present (never was or expired), it should not be displayed as 0 but as `---` indicating that there is no such value available. Especially for depth it would be very dangerous to display and old value when the sensor stopped working and does not send any data. Here 0 is displayed as default, which is obviously not correct.

But in the sail_instrument fields like `nav.gps.sail_instrument.COG` retain their last value when the value has expired and was removed from the store on the server, even though the field no longer contained in the response to `/viewer/avnav_navi.php?request=gps`. :thinking: 

The course=HDM issue was just a side note here.

quantenschaum, 2024-07-04:
The problem with removing expired keys seems to be in `gpsdata.js`. Data is updated and written to the global store in `writeToStore`, the `additionalKeys` are computed from the received data itself, so if a key no longer present in the received data, it is not contained in the `additionalKeys` keys and the old value in the store remains as is.

wellenvogel, 2024-07-07:
Basically gpsdata.js tries to collect all keys in additionalKeys (if they have been received at least once).
This should give exactly what you want - i.e. set the value to undefined in the store if no longer received.
If it has never been received it's not in the store any way.
But maybe there is a bug somewhere...

quantenschaum, 2024-07-07:
Ok, that makes sense, but it doesn't behave like this. - When a field expires on the server it is removed from the store and no longer sent to the frontend. The `additionalKeys` then do no longer contain this key and it doesn't get removed from the store in the frontend.

quantenschaum, 2024-07-07:
The key I am looking at is `nav.gps.sail_instrument.COG`. The problem seems to be that  data structure is not flat, but `additionalKeys['sail_instrument']={...}` contains a dict which is  overwritten in `handleGpsResponse`.

quantenschaum, 2024-08-24:
Related: Position is displayed as `0,0` when there is no fix. The position should be `undefined` instead.

PR #348 (2025-03-14): 347 missing values
Here is a starting point to fix #347

- check for missing values in NMEA parser and omit then (do not set to zero)
- added parsing of `DBK` and `DBS` sentences
- hide course vector if COG is undefined
- hide course vectors in heading mode if zero SOG is detected
- adjusted widgets to display `---` as default if values are undefined
- frontend: merge dicts in state model to collect nested additionalKeys
commit cfb8a86b (2025-03-14): merge dicts to collect nested additionalKeys

fixes #347
closed 2025-03-14 by wellenvogel