wellenvogel/avnav #350: Future Plans for Avnav?

oxbown, 2024-07-09:
First, thank you for the great effort you have put into making Avnav a phenomenal application over the years.
Nowadays almost all navigation applications are paid and even subscription-based. Right now Avnav has become my main navigation app.

Have you considered doing some kind of foundation campaign? For example "Buy Me a Coffee" or "Kickstarter"
Some of the proposals to be developed could be:
- Porting the app to iOS
- Plugins on Android
- Or other major interesting features of the app.

wellenvogel, 2024-07-09:
I already thought a couple of times about this...
But:
I don't think that it is a question of money. For me (fortunately) there is no need to earn money from AvNav. On the other hand I have a lot of fun doing this kind of development.
But I already spend a lot of time in doing it - so even with being able to earn some money my available time would not really increase.
And dealing with money is always a bigger story - with a lot of tasks that do not create that much pleasure...

What I would really appreciate would be some more continous contributions for some of the tasks (maybe not in the core itself but all that is around).

For the proposals:
(1) IOS - for me there is no way in doing that. Would require a third implementation of the server - keeping two of them in sync is already a lot of effort. And the restrictions in the Apple universe are even worse the ones on the google side
(2) plugins on android. I already have some thoughts on that ([ochartsng](https://www.wellenvogel.net/software/avnav/docs/hints/ochartsng.html?lang=en#h2:AndroidApp) is already some kind of plugin). But the plugins will typically not be shared between the linux implementation and the android one (python<-> java|kotlin). But I have some ideas for java script only plugins that could be used in both worlds.

wellenvogel, 2024-12-01:
close this due to no further activity

closed 2024-12-01 by wellenvogel