wellenvogel/avnav #352: Setting time from GPS not working

quantenschaum, 2024-07-19:
Version: 20240708-0907 on OpenPlotter

The options for setting system time from GPS [described in the docs (AVNConfig)](https://www.wellenvogel.net/software/avnav/docs/hints/configfile.html?lang=en) are are not present in the UI.

[image]

When running the Pi offline, the system clock does not seem to set to GPS time.

I noticed that because the track NMEA logs are recorded with wrong time stamps and the track does not get displayed on the map (browser on tablet has correct time).

Am I missing something here or is this bug?

free-x, 2024-07-19:
The heart of OpenPlotter is SignalK. You should rather use the corresponding SignalK plugin
https://github.com/SignalK/set-system-time

free-x, 2024-07-19:
P.S. settime.sh is a part of avnav-raspi package, which is not for OpenPlotter available ( only for AvNav Images)

wellenvogel, 2024-07-19:
Most probably I should be more clear in the docs for that part.
On OpenPlotter we consider SignalK to be the main data handler. 
On AvNav images this is AvNav.
And as OP/SignalK gave their own handling of system time we do not want AvNav to interfere with it.
At the end it's not enough to set the system time - you potentially need to restart some services.
If you still want AvNav to set the system time you can copy the script from the avnav-raspi package and configure it as settime command.

quantenschaum, 2024-07-19:
>configure it as settime command.

OK, but how, the settings are not present? - But I can also let SK to this.

wellenvogel, 2024-07-19:
I would recommend to go with the SignalK approach - it's more natural for OpenPlotter. But at the end I'm not sure if it will handle all services correctly (mdns/avahi does not like setting the system time...).

If you really would like to use AvNav, you must configure the settimecmd at AVNConfig (https://www.wellenvogel.net/software/avnav/docs/hints/configfile.html) - can only be done directly in the xml as it is not intended for "normal" users ;-) -  (you can use the update plugin to edit the xml).
Afterwards you will see all the other configs.
The command from avnav-raspi: https://github.com/wellenvogel/avnav/blob/master/raspberry/settime.sh

quantenschaum, 2024-07-19:
OK, after setting
```xml
<AVNConfig expiryTime="10.0" settimecmd="foo"/>
```
the other config options did appear, the `settimecmd` option itself is not available, though.

BTW, what is avahi used for? Are there devices that auto discover AvNav?

wellenvogel, 2024-07-19:
Yes, settimecmd will never be visible in the UI - as I wrote.
Avahi:
Sure - refer to the docs.
Especially BonjourBrowser.
But also other devices can use this (like my m5 - esp32 stuff).

closed 2024-08-23 by quantenschaum