wellenvogel/avnav #355: Chart detail not showing at various zoom levels

antonical, 2024-07-20:
Hi Andreas,

We have the latest updated UK and Ireland charts via O-Charts. 

It seems that there is an issue when zooming in or out from our home port. See images and what is shown at various zoom levels.

[image]
[image]
[image]

The second image is a useful zoom level but all of the poor information is now shown.

Is this a configuration issue or something we can affect by changing settings?

Cheers
Tony

wellenvogel, 2024-07-20:
Not sure which problem is really the one...
Try to increase overzoom/underzoom.
Such gaps occur if the charts for a particular area differ a lot in their resolution.

antonical, 2024-07-20:
Thats great. I set under zoom to 1 and overzoom to 8 and now have no gaps in the chart. I will experiment with settings to balance a good refresh rate with the details we see.

Thank you for the prompt response and the amazing software.

Cheers
Tony

closed 2024-07-20 by antonical