wellenvogel/avnav #357: Request: Add minute notches to boat course and AIS target vectors

hkapanen, 2024-07-30:
I also suggest adding "minute notches" to boat course and AIS vectors similar to what [iSailor](https://images.yachtingjournal.com/images/maritime/image-transas-47693.png) has. This feature has proven to be very helpful on predicting where the boat or AIS target will be in a few minutes.

wellenvogel, 2024-11-30:
Sorry for the long delay - was busy with other stuff.
Interesting idea.
I'm not sure how much performance this will cost - it's at least a lot more drawing that would be necessary (especially for a large number of AIS targets).
Not sure if it would already help to only get this for the boat itself.

hkapanen, 2024-12-10:
Boat alone already has a benefit, but also closest AIS targets have a huge benefit when considering CPA visually.
To address possible performance concerns with large number of AIS targets, consider doing this only for a small number (10?) of the closest targets.

quantenschaum, 2026-08-18:
Good idea. For visualizing CPA use the [relative motion vectors](https://www.wellenvogel.net/software/avnav/docs/userdoc/navpage.html?lang=en#h3:AISMotionVectors).