wellenvogel/avnav #359: 'Housekeeping' Tracks

nmeyne, 2024-08-01:
We are a club of skippers who share an old boat.  It is nice to clear up the tracks (and any temporary routes) for the next skipper.   It is possible to clear current and previous tracks by deleting the .gpx for the relevant days, but less confident users are worried about deleting the wrong thing (eg a chart)  in the files menu. So:

Some simple buttons / dialog to empty (and restore) the current track would be helpful.

wellenvogel, 2024-08-02:
Interesting idea.
Need to think about a bit. Currently there is no easy concept of doing such things, Basically it would require to store some data separately with the option to re-import them.
What could be fairly easy:
Just empty the current track (but keep the existing data as a file with some extended name - like adding a time stamp to the day in the name).

nmeyne, 2024-08-02:
`Just empty the current track (but keep the existing data as a file with some extended name - like adding a time stamp to the day in the name)`

Great. The restore isn't important if the original data is saved.   Some extra documentation about the track behaviours and naming would be fine.

nmeyne, 2024-08-02:
It could also be useful to store the track distance in the name of the old track as well as the time it was closed.  This could provide some basic trip log function, perhaps?  [See issue 361](https://github.com/wellenvogel/avnav/issues/361)