wellenvogel/avnav #361: 'Trip Log' Functionality

nmeyne, 2024-08-02:
AvNav does not have a  'trip log' widget / display.  There is no persistent record of total distance run over the ground (or through the water).  Normally this would be built into a log device with some persistent storage of some kind, but not in a 'headless' GPS that sends NMEA only.   Also AvNav currently does not have a data base that survives e.g. restarts of the system.

Distance run could be obtained (approximately) from the length a track by summing up distances between points with some parameter to ignore/suppress position errors .

The persistent storage issue could be overcome by doing the calculation when the track is closed or cleared and then including the distance run in the .gpx filename along with the time closed/cleared.

The filenames in the tracks list would then provide a handy list of distances run for each track.

Does that  make sense?

nmeyne, 2024-08-02:
Sorry, I have just realised that the [Track 'Info' dialog](https://www.wellenvogel.net/software/avnav/docs/userdoc/downloadpage.html) already provides the track length information and more.
I guess it still might be helpful to see it in the filename or in an overview on the list of tracks?

wellenvogel, 2024-08-02:
Seeing it in the filename could be interesting. But until now the computation is done at the browser side (not on the server). So there is no chance to do this when closing the file (until this would be re-implemented on the server side - for linux and for android).
Showing it in the overview could be a problem as tracks could contain many points - and computing the distance on the fly for a lot of tracks could slow down things a lot.

nmeyne, 2024-08-02:
Thanks, yes I guess the computation would need to be moved server-side.  At least it would run infrequently there, on file close.  Doing it client side on the fly for many large tracks indeed sounds very slow. I guess this is a bigger job than expected!