wellenvogel/avnav #362: Additional raster chart source: VMH

nmeyne, 2024-08-06:
Several of our skippers have a preference for a raster 'paper chart' look and feel despite the big steps between zoom levels and chart areas.  In the UK, VMH 'visit my harbour' team sell raster chart sets that seem well priced.  Having a choice between o-charts oeSENC vector charts and a set of raster charts would be very welcome for us.

VMH seem to be open to new implementations of their charts on multiple platforms (most recently via [qtVlm](https://www.meltemus.com/index.php/en/)).  They are BSB / .KAP files I think, with activation tied to a specific machine and activation code, possibly similar to the o-charts activation.   Do you think it is feasible/worthwhile to implement?

free-x, 2024-08-07:
Did you tried following steps
- Install AvNav for Windows (  http://wellenvogel.net/software/avnav/docs/install.html?lang=en#h2:Windows)
- Mount and unlock VMH ISO (https://www.visitmyharbour.com/articles/4386/set-up-the-unified-download-iso-for-easy-opencpn-use)
- Try to convert the charts from drive V:

Be careful, there are many charts and resulting GEMF can be very large

wellenvogel, 2024-08-07:
Interesting idea...
When using the converter you will most probably need to use the link feature (https://wellenvogel.de/software/avnav/docs/userdoc/importerpage.html?lang=en#h2:CopyingFilesExperts).
So you need to create a file with the .clk extension and one line with the path to the chart directory inside.
Not sure if uploading the charts via the webbrowser would work.

nmeyne, 2024-08-07:
@wellenvogel @free-x Many thanks. Worth a try.... Back to Windows! :-/

nmeyne, 2024-08-08:
> Try to convert the charts in AvNav from drive V:

I get errors from the converter:
```
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "C:\Users\nick\AppData\Local\avnav\server\handler\../..\chartconvert\read_charts.py", line 1827, in <module>
main(sys.argv)
File "C:\Users\nick\AppData\Local\avnav\server\handler\../..\chartconvert\read_charts.py", line 1819, in main
raise Exception(strerr)
Exception: ERROR: unable to copy/replace C:\Users\nick\AvNav\work\UKandIreland\work\UKandIreland.gemf\tmp.gemf to C:\Users\nick\AvNav\charts\UKandIreland.gemf: Traceback (most recent call last):
File "C:\Users\nick\AppData\Local\avnav\server\handler\../..\chartconvert\read_charts.py", line 1814, in main
shutil.copy(tmpgemf,outtmp)
File "D:\obj\windows-release\37amd64_Release\msi_python\zip_amd64\shutil.py", line 248, in copy
File "D:\obj\windows-release\37amd64_Release\msi_python\zip_amd64\shutil.py", line 120, in copyfile
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\nick\\AvNav\\work\\UKandIreland\\work\\UKandIreland.gemf\\tmp.gemf'

2024/08/08-18:12:29 ERROR: conversion failed
```
The ISO is mounted as drive D with OpenCPN 'portable' executing from there.  KAP charts look fine in Open CPN

wellenvogel, 2024-08-08:
Can you please attach the full log?

nmeyne, 2024-08-08:
[convert-0030-0.log](https://github.com/user-attachments/files/16554016/convert-0030-0.log)
Here is a full one, sorry.

free-x, 2024-08-09:
Do I understand correctly that the "kap" files were copied to the import directory? 
I don't quite understand the DRM protection of the VMH. Can you provide a sample file of VMH kap?

wellenvogel, 2024-08-09:
Please follow my advice for using the link feature.
Copying the .kap files to the import dir or uploading them could potentially not work.
In the log you can see that the .kap file cannot be read.

nmeyne, 2024-08-09:
Thanks.  Yes I think copying does not work.  Sadly, adding the .clk link file for the whole directory, followed by a rescan did not seem to convert the files either.  Example:

> Exception: Invalid file: D:\UK-IRL-2024\3741-1.KAP
> converting D:\UK-IRL-2024\3741-1.KAP to C:\Users\nick\AvNav\work\UK-IRL-2024\work\UK-IRL-2024.gemf\basetiles\3741-1.vrt failed - trying to use native
> ERROR 4: `D:\UK-IRL-2024\3741-1.KAP' not recognized as a supported file format.
> gdal cannot handle file D:\UK-IRL-2024\3741-1.KAP

Beginning of the log attached:
[convert-UK-IRL-2024.log](https://github.com/user-attachments/files/16559162/convert-UK-IRL-2024.log)

nmeyne, 2024-08-09:
> Do I understand correctly that the "kap" files were copied to the import directory? I don't quite understand the DRM protection of the VMH. Can you provide a sample file of VMH kap?

The DRM prevents copying from the ISO, but seems to allow direct access.  I have an evaluation copy which I'm trying to use with AvNav, but I think copying to work directories for processing is prevented, as is any protected file/directory copying on the host machine.  OpenCPN, however, can access and render the chart files directly.  This works for both the evaluation CPN copy on the ISO and for a full install of OpenCPN assigned/pointed to the ISO chart directory [image]  If I can get it to work with AvNav, I'd be very happy to buy a full chart set, and subsequent updates, but at the moment I have only purchased the evaluation copy for 14 days.   See the conditions statement attached [image]  .  The statement "other programs and apps are supported in the full downloadable version of Unified Charts"  (Windows only) suggests that the program may have to be registered with the DRM to enable access, perhaps?

free-x, 2024-08-09:
I think, at first, you can expect KAPs with gdalinfo. 
Could you extract image from KAP with imgkap? (https://github.com/nohal/imgkap/releases/tag/v1.16.2)

BTW. i once found a link to buy trial and now it is lost again

nmeyne, 2024-08-09:
I could not get imgkap to run - missing VCOMP120.DLL
Here is the link to the trial: https://www.visitmyharbour.com/articles/4094/unified-charts-by-downloadtest-compatibility-now

free-x, 2024-08-09:
DLL is a part of M$ VC++ redistributable package
https://support.microsoft.com/en-us/topic/update-for-visual-c-2013-redistributable-package-d8ccd6a5-4e26-c290-517b-8da6cfdf4f10

P.S. I can't find where I can pay £1.95. Crap

nmeyne, 2024-08-09:
Thanks for the  DLL, imgkap is a great tool, but:
```
C:\Users\nick\Downloads\imgkap>imgkap D:UK-IRL-2024\0002-0.KAP C:\Users\nick\Downloads\0002-0.png
ERROR - Invalid index table in (null)
ERROR - imgkap ((null)) return 3
```
VMH Payments / codes
I called them yesterday after a lot of trouble with their online shop payment page.  Eventually the checkout worked and I got an email.  The code didn't work.  They gave me an activation code over the phone and that worked OK.
The page I looked at and paid from seems to have been taken down? https://www.visitmyharbour.com/articles/4473/try-unified-charts-on-windows-for-14-days

Not great.

nmeyne, 2024-08-09:
To implement another DRM in AvNav doesn't sound like much fun (!) unless there is perhaps some re-use possible from what VMH have done with qtVlm.   VMH distribute the charts for qtVlm in mbtiles I think, with another DRM activation tool on it:  see this pic from their forum:
[image]

This seems like a big job?   In the interim I guess it might be possible to manually create a few 'favourite chartlets' from the chartset with some OpenCPN screen grabs and use those as overlays in AvNav?

free-x, 2024-08-11:
the last chance. Rename imgkap.exe to opencpn.exe and try again
For example:
```
C:\Users\nick\Downloads\imgkap>opencpn.exe D:UK-IRL-2024\0002-0.KAP C:\Users\nick\Downloads\0002-0.png
```

nmeyne, 2024-08-12:
@free-x No, same result.

@wellenvogel I think the team at VMH are not interested in enabling their charts for AvNav or creating a plugin, so I will close this issue now.

closed 2024-08-12 by nmeyne