wellenvogel/avnav #363: Request: Show "route distance" and ETA to any waypoint

AaskovKnudsen, 2024-08-09:
Thank you so much for AVNAV. I really like it.
Allow me a feature request:
For now, you have "route distance" and ETA to next wp and to end of route.
It would be very usefull to know the "route distance" and ETA to a certain waypoint, for instance to reach a bridge opening.
Could for example be implemented on the route screen: Seleting a wp could show a dialog with these informations, and perhabs the buttons "show wp" and "cancel"

wellenvogel, 2024-08-10:
Looks like an interesting feature.
The only problem I see is the workflow for setting/changing this point.
And we would need a nice naming for this "special route point".
Most probably the easiest approach from the user perspective would be to allow to select one of the points of a route as this special point. And then there could be a widget to show information regarding this point.
But this would require some extension in the storage format of the route (non standard).

Another option would be to only "dynamically" select such a point when a route is active (could become part of the feature info dialog when tapping this point).
Drawback: Would most probably not be persistent (i.e. only available as long as the route is active).
But maybe this could be combined with a more general feature: Show target information (eta/dst/brg...) to an arbitrary point beside the current routing target.
I guess this second option would be more easy to implement.

AaskovKnudsen, 2024-08-14:
Hi Andreas, and thank you for your answer.
I agree in the second option. This feature only makes sense if you have an active route.
As I also read in your answer, it could be as simple as selecting a wp from the wp-list in the route page.
This should show a dialog with distance from boat (following the route) and ETA to the selected wp, and perhabs the buttons "show wp" and "cancel".
Of course the information will only be a snapshot, and only updated by selecting the wp again.
This will be sufficient for me, and hopefully not so hard to implement. 

[image]

wellenvogel, 2024-08-15:
Hmm - not exactly what I thought . And it would not easily address your use case.
I also had this issue (reaching a bridge opening) many times (basically every time I returned to my home port in Greifswald) - and I always needed to have a display of the ETA all the time (I always had this as the last point in the route).
So I guess to really address the use case you would need the option for a permanent display of those data for a selected point (similar to the next way point display(s)).
Therefore my idea was to add a feature like "secondary target point" - and provide widgets that would display dst,eta,brg,... to this point - either directly or on the current route. 
The easiest implementation would just be integrated with the feature info on the navigation page (i.e. the dialog that pops up whenever you click on the map) - similar to the "GoTo" button in this dialog. Something like "2nd target" (but I would love to have a better name for it).
Maybe the information could also be added to the feature info dialog already.

For me it does not make that much sense to show this in the route editor...

AaskovKnudsen, 2024-08-16:
Yes, it is much better.
Perhaps you could use the "ActiveRoute" widget for both the last point and the selected point. (last point as default)
Click the widget, and select the point you want from a list.
In iSailor it is done like this.