wellenvogel/avnav #365: Can Avnav send the complete route out via nmea?

antonical, 2024-08-14:
Can avnav send the route information out over nmea?

RTE - RTE is sent to indicate the names of the waypoints used in an active route. There are two types of RTE sentences. This route sentence can list all of the waypoints in the entire route or it can list only those still ahead. Because an NMEA sentence is limited to 80 characters there may need to be multiple sentences to identify all of the waypoints. The data about the waypoints themselves will be sent in subsequent WPL sentences which will be sent in future cycles of the NMEA data.

  $GPRTE,2,1,c,0,W3IWI,DRIVWY,32CEDR,32-29,32BKLD,32-I95,32-US1,BW-32,BW-198*69
           RTE          Waypoints in active route
           2            total number of sentences needed for full data
           1            this is sentence 1 of 2
           c            Type c = complete list of waypoints in this route
                        w = first listed waypoint is start of current leg
           0            Route identifier
           W3IWI,...    Waypoint identifiers (names)
           *69          checksum
           
On our yacht we have a combination of technologies. At the chart table we have a low power 12v nav computer running ubuntu 22.04 and headless avnav server running on a rpi4.

We access avnav by a browser at the nav computer. Also at the chart table we have a dedicated Raymarine plotter and also one in the cockpit. 

We do all of our route planning on avnav. We send the navigation information from avnav to the boat seatalk network and it is received correctly at the Raymarine MFD's and the ST6001+ autopilot but I believe it is only the active waypoint. But it would be great to send the entire route to the raymarine MFD's before we start navigating so we have the same info on all devices. We have the same navigation data (pos, wind, depth etc. etc.) as that is on the network.

We can also consume the avnav navigation data via phone and tablet. as everything apart from the dedicated raymarine kit is on the yachts ethernet and wifi network.

wellenvogel, 2024-08-15:
For sure this would be an interesting feature.
But there are a couple of problems for me:
(1) no clear standard (especially about the naming restrictions for the points in the route - AvNav basically has none there)
(2) at my side a lack of devices that could be used for testing.

antonical, 2024-08-15:
Many thanks. I was thinking from the routes page. We could select a route and then send out via (select a configured connection).

Cheers