wellenvogel/avnav #366: Navigation map twitching

gtf375, 2024-08-15:
When navigating north (approximately 352° to 8°) and heading on course, the map appears to twitch along with the route and vessel icon. When heading orientation is turned off, the twitching stops. Version 20040704 (in previous versions there was also twitching), S57 maps, true HDT direction of movement from the GPS compass using nmea2000 (when using COG, it also twitches). I have never seen such a problem described anywhere. Probably the problem is in the settings, but I can’t figure out where to look.

https://github.com/user-attachments/assets/38794acf-d7aa-4e0d-ac19-9634ba646e97

wellenvogel, 2024-08-15:
Could you provide a log of the NMEA data you have?
Maybe there the smoothing function is not working correctly.
You can try to play with settings/navigation/Rotation Tolerance.

wellenvogel, 2025-01-10:
Could reproduce this - was caused by OL trying to constrain the view rotation around 0.

wellenvogel, 2025-07-23:
release with [20250723](https://www.wellenvogel.de/software/avnav/downloads/release/20250723/?lang=en)

closed 2025-07-23 by wellenvogel