wellenvogel/avnav #367: True Wind Data in NMEA

quantenschaum, 2024-08-15:
I recently had some exchange with the guys from https://www.yachtd.com/ about the calculation of true wind from apparent wind and course data in one of their products. 

>me: this means that it is not possible to send true wind data using NMEA0183 at all?  
>they: Formally, yes, as there is simply no clear definition in NMEA 0183 what the True wind is.

Now I am wondering how to correctly read/write true wind data from/to NMEA0183.  
It is less obvious than one might think at first glance.

## Definitions

- **apparent** wind = wind as measured on the ship
- **true** wind = wind relative to (moving) water as observed when _not_ moving through water
- **ground** wind = wind relative to ground as observed when _not_ moving over ground
- wind **direction** = direction of the wind relative to (true) north
- wind **angle** = angle of the wind relative to (true) heading (ship's center line)

## Sentences

NMEA [sentences containing wind data](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:nmea0183)

- [MWD](https://gpsd.gitlab.io/gpsd/NMEA.html#_mwd_wind_direction_speed) - Wind Direction & Speed
- [MWV](https://gpsd.gitlab.io/gpsd/NMEA.html#_mwv_wind_speed_and_angle) - Wind Speed and Angle
- [VWR](https://gpsd.gitlab.io/gpsd/NMEA.html#_vwr_relative_wind_speed_and_angle) - Relative Wind Speed and Angle
- VWT - True Wind Speed and Angle (obsolete)

Let's have a look at these.

MWD contains wind direction (rel. to true/mag north) and wind speed.  
**But what wind, apparent or true or ground?**  
We may assume _true wind_, here but the docs are not clear.

MWV contains wind angle (rel. to heading) and wind speed.  
**But again what wind, apparent or true or ground?**  
We may assume _apparent wind_ here, but the docs are not clear.

The [docs on gpsd](https://gpsd.gitlab.io/gpsd/NMEA.html#_mwv_wind_speed_and_angle) seem to be wrong about the reference flag, the write

>Reference, R = Relative, T = True

But T actually stands for "theoretical" ([compare to this](https://docs.vaisala.com/r/M211109EN-L/en-US/GUID-7402DEF8-5E82-446F-B63E-998F49F3D743)). From the yachtd folks I got this description (source not known to me).

[image]

This means MWV data with the R flag = apparent wind, with T flag it could be _true_ or _ground_ wind, I cannot really tell from the description.

Furthermore there is VWT, which does contain true wind (rel. to water), but the use of this sentence is discouraged (why?). One should use MWV instead, which is not well defined. So, one may conclude that a T flagged MWV does contain true wind.

[image]

VWR contains apparent (relative) wind.

## Summary

Despite the unclear definitions in the docs I would assume

- MWD - true wind direction and speed
- MWV (R) - apparent wind angle and speed
- MWV (T) - true (theoretical/calculated) wind angle and speed
- VWR - apparent wind angle and speed (as MWV (R))
- VWT - true wind angle and speed (as MWV (T))

wellenvogel, 2024-08-15:
Thanks for digging into those details.

quantenschaum, 2024-08-21:
For completeness the N2K PGNs for wind data.

Wind Data is sent in [PGN 130306](https://ttlappalainen.github.io/NMEA2000/group__group__msg_set_up.html#ga5dca3e1aed795b9f865a955bf4182df3) which contains an angle/direction and a speed. There the following [WindReferences](https://ttlappalainen.github.io/NMEA2000/group__group__msg_types.html#ga7e2df5fcd75db7834060bc276d1b741f) defined.

- `True_North` = Ground Wind **direction** (rel. to true north) and speed
- `Magnetic` = Ground Wind **direction** (rel. to mag. north) and speed
- `Apparent` = Apparent Wind **angle** (rel. to boat center line) and speed
- `True_boat` = Ground Wind **angle** (rel. to boat center line) and speed
- `True_water` = True Wind **angle** (rel. to boat center line) and speed

quantenschaum, 2024-08-22:
for the record

In the [manual](https://www.yachtd.com/downloads/ydwg02.pdf) of the [Yacht Devices WiFi Gateway YDWG-02](https://www.yachtd.com/products/wifi_gateway.html) and others they write (page 32)

>Historically, STW/HDG are used to calculate true wind. However, this is not correct
in places with strong current, and the «true» value of true wind can be obtained using
the SOG/HDG pair. Therefore, our gateways offer four options: SOG/HDG (if you love
truth), SOG/COG (if you have GPS only), STW/HDG (if tradition is most important),
or you can disable calculations. In the last case the gateway reports true wind data
only if they are calculated by another device available on NMEA 2000 network.

IMHO this statement is totally  wrong! 

- True wind _must_ be calculated using STW/HDG (and leeway) otherwise it's not true wind.
- Wind calculated from SOG/HDG is something completely undefined, a crude mixture of true and ground wind. 
- Wind calculated from SOG/COG (and HDG to convert AWA->AWD) is ground wind.

:heart: open source software where you can actually see how it is done in the code and even change it!

wellenvogel, 2025-03-24:
As far as I see the code inside AvNav should be correct right now - what do you think?
So I will finally try to integrate some of those hints into the doc.

quantenschaum, 2025-03-26:
Let' see...
- NMEA 18xx parsing
  - MWV ▶ TWA/TWS if `ref=='T'` else AWA/AWS ✅ 
  - MWD ▶ TWD/TWS ✅ 
  - VWR ▶ AWA/AWS ✅ 
  - VWT is not parsed
- NMEA2k is not handled directly, right?
- SignalK: the keys from `avnav_nmea.py` are used in `signalkhandler.py` ✅ 
  - https://github.com/SignalK/specification/issues/520 is not yet resolved, so I cannot tell what the correct keys in SK are.

isarychkin, 2025-06-01:
> for the record
/////
> 
> * True wind _must_ be calculated using STW/HDG (and leeway) otherwise it's not true wind.
> * Wind calculated from SOG/HDG is something completely undefined, a crude mixture of true and ground wind. (Might be used as fallback when there is no compass or no log, but the result is wrong.)
> * Wind calculated from SOG/COG (and HDG to convert AWA->AWD) is ground wind.
> 
interesting that NKE actually suggest using SOG/HDG for cruising and autopilot mode, substituting boat speed reference for SOG in their instruments settings: "... this calculation based on bottom speed is useful for the pilot but not for measuring performance. When cruising it is also a very interesting speed reference." (https://nke-marine-electronics.com/faq/)
It's good to be able to choose what's more appropriate for your specific needs.
Regards