wellenvogel/avnav #368: outdated docs

quantenschaum, 2024-08-16:
I just tried to install the AvNav image for the PI and followed the instructions on https://www.wellenvogel.net/software/avnav/docs/install.html

The docs seem to be outdated, links are dead or pointing to old resources.

- current image is from 2023
- user "blacksea" yields 404
- https://github.com/free-x/avnav is not helpful

This quickly gets very confusing and frustrating for new users.

free-x, 2024-08-16:
Hi,

current **stable** image is indeed from 2023
There is also [experimental Image](https://www.segeln-forum.de/thread/91582-beta-avnav-image-20240605/) 
Link to "BlackSea" profile at segeln-forum.de is broken( seems were changed by forum software). Correct: https://www.segeln-forum.de/cms/user/27970-blacksea/ 
BTW: it's me ;-)

> https://github.com/free-x/avnav is not helpful
I can't find this link at @wellenvogel 's page

Regards
free-x

quantenschaum, 2024-08-16:
The Link to your avnav repo is behind "seiner Webseite" (just after BlackSea).

quantenschaum, 2024-08-16:
Maybe the bullseye image should be updated to contain recent versions.

wellenvogel, 2024-08-17:
more or less duplicate to #345 for the docs.
Will try to get those corrected with the next release.
Most probably we will also build a new image. Currently I'm struggling with updating to reactjs 18.

quantenschaum, 2024-08-25:
The releases listed here

- https://www.wellenvogel.net//software/avnav/docs/release.html and
- https://www.wellenvogel.net//software/avnav/downloads/release/

do not seem to match what the auto updater in AVNav itself offers.