wellenvogel/avnav #373: AIS Targets additional information lines

c-graeser, 2024-09-04:
Chosing CPA as second line of information beside Name/MMSI the lines mix up under the condition when the vessel sends COG 360 degrees.
This seems to happen frequently when moored.
[image]

wellenvogel, 2024-09-12:
Do you have any test data available that contains this situation?
You could just create an NMEA log with all filters removed at the logger and the interval set to 0.

c-graeser, 2024-09-13:
Hello,
attached you find AIS Data.
[20240913-AIS.txt](https://github.com/user-attachments/files/16990066/20240913-AIS.txt)
Hope it helps.

c-graeser, 2024-09-13:
Maybe stupid question:
How can I test your changes?
I have no file "aislayer.js" in my raspberry system.
Is it somhow compiled?

c-graeser, 2024-09-13:
Here is mor logfile with all data including GPS
[2024-09-13-AIS-v2.txt](https://github.com/user-attachments/files/16990538/2024-09-13-AIS-v2.txt)

wellenvogel, 2024-09-13:
> I have no file "aislayer.js" in my raspberry system.
Is it somhow compiled?

Yes it's compiled. So you would need a nodejs environment (or just Java and start with the global gradle build).
But it requires some skills...
So the easiest solution will be to wait for the next daily build.

c-graeser, 2024-09-20:
Super Arbeit!
Jetzt werden die Daten der AIS Ziele korrekt dargestellt.
Vielen lieben Dank!

wellenvogel, 2025-07-23:
release with [20250723](https://www.wellenvogel.de/software/avnav/downloads/release/20250723/?lang=en)

closed 2025-07-23 by wellenvogel