wellenvogel/avnav #374: Clinometer in Android App

bk138, 2024-09-23:
@wellenvogel would there be interest from your side to merge support for a clinometer backend and widget to the Android app? Would be very handy for a sailing vessel.  I want to ask in advance before starting any work because:

- This only makes sense if the Android device is mounted to the sailing vessel in a fixed position. Probably self-explanatory to users.
- The instrument screens 1,2,3 in avnav are all already "full". This would need to be taken care of somehow.

If you're generally positive, I'd prepare a PR with the needed sensor changes and a very simple digit display (just a numeric degree view for now, indicating heel left/right or port/starboard).

Let me know what you think!

bk138, 2024-09-23:
Sorry, just searched for "clinometer" in existing issues, now found #101 on a 2nd look. Feel free to close either as a dupe.

wellenvogel, 2024-09-24:
Could be an interesting feature.
The display is no problem at all - https://www.wellenvogel.net/software/avnav/docs/hints/layouts.html?lang=en.
So for a basic display nothing is to be done on the Browser side.
One could also use the SKPitchWidget to display the value (it can handle input in deg).
For the (Android-) server part:
Currently there is no real infrastructure to handle sensors.
Most probably a new Worker would be appropriate. Maybe it could be prepared to handle more sensors.
And it should be able to output a NMEA record (XDR) with the pitch value (or whatever it detects). 
This way the decoder will automatically pick up the value and will provide it to the UI.
Potentially similar to what  https://github.com/wellenvogel/avnav/blob/master/server/handler/sensehat.py is doing for the linux part.
And I guess the worker would need some config parameter(s) to somehow calibrate/adjust/enable/disable.
If you would like to start some work on that I would appreciate it.