wellenvogel/avnav #375: Show position of AIS-vessel

AaskovKnudsen, 2024-09-24:
Is there a way to show a selected ais-vessel on the map?
If I select a vessel from the ais-list, I get the ais-info for that vessel. This is fine.
Pressing the green triangle button in that window, I would expect the selected ais-vessel to be centered on the map.
Instead my own boat is centered on the map.
Is that the intended behavior?

Also, a nice feature would be to be able to sort the ais-list by name.

wellenvogel, 2024-09-24:
If you press the center button on the info page you get what you want.
Pressing the green triangle button means "back to normal" -i.e. showing the nearest target in the AIS widget (and center to this one).

Sorting by name could be a nice feature (although it's hard for me to imagine the use case). Currently you could at least filter by name....
But we can leave this issue open to include a sort-by-name feature.

AaskovKnudsen, 2024-09-25:
Thank you for clarifying that. I should have seen that myself.

A sort-by name feature could be usefull if you are looking for "friends" in the area.