wellenvogel/avnav #376: Plugins do not start when run on python 3.12

olivier-van-wijngaarden, 2024-10-11:
Recently I upgrade my laptop to Ubuntu 24.4 .  The  default python  version I have now is 3.12. Since the upgrade AvNav failed to start.
The root cause seemed to be in pluginhandler.py. This uses the python imp module. This module was already deprecated a few versions ago, but since 3.12 it have been removed. From the imp module it uses only load_sources().
I replaced it as decribred below (inspired from https://devcodef1.com/news/1027645/replacing-imp-load-source-in-python-3-12), which worked for me:

< import imp
---
> import importlib
657c657,659
<       rt = imp.load_source(name, moduleFile)
---
>       spec = importlib.util.spec_from_file_location(name, moduleFile)
>       module = importlib.util.module_from_spec(spec)
>       spec.loader.exec_module(module)
659c661
<       return rt
---
>       return module
****

wellenvogel, 2024-10-11:
Thanks a lot for bringing this up.
We did not test on ubuntu 24.4 until now - but having a more future proof solution will be great.

wellenvogel, 2025-07-23:
release with [20250723](https://www.wellenvogel.de/software/avnav/downloads/release/20250723/?lang=en)