wellenvogel/avnav #379: Ocharts disappeared after update to 240616

Fietsenwiel, 2024-10-21:
After I updated Avnav from 20230705 to 240616 the Ocharts are disappeared.
I did the update from within AvNav.

The version of the avnav-charts packages is 1.0.34.0-1bullseye
The version of the avnav-ocharts-plugin package is 20230706-raspbian-bullseye

I really like to update avnav but I also like to keep my Ocharts.
What is going wrong?

free-x, 2024-10-21:
Don't know, what's going wrong
For me works with AvNav 20240914 (daily) + AvNav-ocharts-plugin (20231216-raspbian-bullseye) 

Could you provide your logs? ( avnav/data/log/avnav.log  and avnav/data/ocharts/provider.log)

Greetings
free-x

Fietsenwiel, 2024-10-21:
I have two logfiles, one of version 20230705 and one of version 20240616...
[avnav-20241021 version 20230705.log](https://github.com/user-attachments/files/17461717/avnav-20241021.version.20230705.log)

[avnav-20241021 version 2024061.log](https://github.com/user-attachments/files/17461752/avnav-20241021.version.2024061.log)

wellenvogel, 2024-10-21:
Can you please describe in detail how you started AvNav.
In the newer version you run AvNav from a different directory. This could happen if you start AvNav by hand...
On this other directory (/var/lib/avnav) none of your data is available.

Fietsenwiel, 2024-10-21:
Do you mean this?:
python3 /usr/lib/avnav/server/avnav_server.py -q -l INFO -c /var/lib/avnav/charts -w /var/lib/avnav -u viewer=/usr/lib/avnav/viewer /var/lib/avnav/avnav_server.xml

And this?:
/usr/lib/avnav/server/../plugins/ocharts/bin -w 8000 -n /usr/lib/avnav/server/../plugins/ocharts/lib/opencpn /usr/lib/avnav/server/../plugins/ocharts/share/opencpn /var/lib/avnav/ocharts 8082

wellenvogel, 2024-10-21:
no, could you just answer how you started AvNav?
All the time normally "automatic" start by the service?
Or any start from the command line?

wellenvogel, 2024-10-21:
Could you please execute the command
`sudo systemctl status avnav`
When running the updated version (i.e. 20240616) and paste the output here.

Fietsenwiel, 2024-10-22:
AvNav starts automatic when the Raspberry pi boots.

`pi@Relexed:~ $ sudo systemctl status avnav
* avnav.service - NMEA multiplexer and Browser based navigation
     Loaded: loaded (/lib/systemd/system/avnav.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2024-10-21 15:34:35 CEST; 16h ago
   Main PID: 5902 (python3)
      Tasks: 78 (limit: 3933)
        CPU: 39min 56.472s
     CGroup: /system.slice/avnav.service
             |-5902 python3 /usr/lib/avnav/server/avnav_server.py -q -l INFO -c /var/lib/avnav/charts -w /var/lib/avnav -u viewer=/usr/lib/avnav/viewer /var>
             |-5961 /bin/sh /usr/bin/xvfb-run -n 1000 -s -screen 0 512x512x24 /usr/lib/avnav/plugins/ocharts/AvnavOchartsProvider -t 5 -d 1 -s 2 -l /var/lib>
             |-5974 Xvfb :1000 -screen 0 512x512x24 -nolisten tcp -auth /tmp/xvfb-run.TmnRza/Xauthority
             |-5989 /usr/lib/avnav/plugins/ocharts/AvnavOchartsProvider -t 5 -d 1 -s 2 -l /var/lib/avnav/ocharts/provider.log -p 5902 -c 10000 -f 400000 -r >
             |-6013 dbus-launch --autolaunch 0af77b5891a544129c09386b8be25e74 --binary-syntax --close-stderr
             `-6014 /usr/bin/dbus-daemon --syslog-only --fork --print-pid 5 --print-address 7 --session

okt 21 15:34:35 Relexed systemd[1]: Started NMEA multiplexer and Browser based navigation.
okt 21 15:34:35 Relexed avnav[5902]: chartdir=/var/lib/avnav/charts, basedir=/var/lib/avnav, config=/var/lib/avnav/avnav_server.xml, starting server...
okt 21 15:34:37 Relexed avnav[5902]: 2024-10-21 15:34:37,491-5902-5902-MainThread-INFO-basedir=/usr/lib/avnav/server,datadir=/var/lib/avnav
okt 21 15:34:37 Relexed avnav[5902]: 2024-10-21 15:34:37,491-5902-5902-MainThread-INFO-####start processing (version=20240616, logging to /var/lib/avnav/log>

`

free-x, 2024-10-22:
Which flavor do you use? Seems to be neither AvNav Image nor Openplotter
I'm missing systemd overriding for AvNav
```
User=pi
ExecStart=/usr/bin/avnav -q -b /home/pi/avnav/data -t /usr/lib/avnav/raspberry/avnav_server.xml -n /etc/default/avnav
```

free-x, 2024-10-22:
> sudo nano /usr/lib/systemd/system/avnav.service.d

It's wrong
Correctly
```
sudo mkdir /usr/lib/systemd/system/avnav.service.d
sudo nano /usr/lib/systemd/system/avnav.service.d/my.conf
```

Fietsenwiel, 2024-10-22:
Ok, but it worked with previous version 20230705.

I'll try your correction tomorrow and report the result.

Thanks in advance.

wellenvogel, 2024-10-22:
Hmm - you did not really follow the instructions:
Refer to https://wellenvogel.de/software/avnav/docs/install.html?lang=en#h2:PackageInstallation

```Create /usr/lib/systemd/system/avnav.service.d and create a file avnav.conf in this directory with the following content...
```

Fietsenwiel, 2024-10-22:
I don't understand your solution.

/usr/lib/systemd/system/avnav.service.d already exists but it is not a directory:

pi@Relexed:/ $ ls -al /usr/lib/systemd/system/avnav.service.d
-rw-r--r-- 1 root root 203 15 apr  2024 /usr/lib/systemd/system/avnav.service.d

And I have more then one avnav.conf:
pi@Relexed:/ $ sudo find * | grep avnav.conf | grep -v geweigerd
home/pi/avnav/data/ocharts/avnav.conf
usr/lib/avnav/plugins/ocharts/avnav.conf
var/lib/avnav/ocharts/avnav.conf

They are all different:
pi@Relexed:/ $ ls -al  var/lib/avnav/ocharts/avnav.conf
-rw-r--r-- 1 avnav avnav **3261** 22 okt 07:28 var/lib/avnav/ocharts/avnav.conf
pi@Relexed:/ $ ls -al usr/lib/avnav/plugins/ocharts/avnav.conf
-rw-rw-r-- 1 root root **3252**  6 jun  2020 usr/lib/avnav/plugins/ocharts/avnav.conf
pi@Relexed:/ $ ls -al home/pi/avnav/data/ocharts/avnav.conf
-rwxrwxrwx 1 avnav avnav **631** 21 okt 12:23 home/pi/avnav/data/ocharts/avnav.conf

wellenvogel, 2024-10-22:
You simply need to exactly follow the documentation.
Just remove the file that you created wrongly:
/usr/lib/systemd/system/avnav.service.d
And afterwards create a directory with this name and place a file avnav.conf into it with the described content.
All the other avnav.conf have their own meaning - and are not related to the problem we discuss here.
You could have a look at the systemd documentation if you would like to understand the rules.

Fietsenwiel, 2024-10-22:
I did the following:

pi@Relexed:~ $ rm /usr/lib/systemd/system/avnav.service.d
pi@Relexed:~ $ mkdir /usr/lib/systemd/system/avnav.service.d
pi@Relexed:~ $ cd /usr/lib/systemd/system/avnav.service.d
pi@Relexed:/usr/lib/systemd/system/avnav.service.d $ nano avnav.conf

#Overrides for the avnav service
[Service]
User=pi
ExecStart=
ExecStart=/usr/bin/avnav -q -b /home/pi/avnav/data -t /usr/lib/avnav/avnav_template.xml -n /etc/default/avnav

[Unit]
After=avnav-check-parts.service

pi@Relexed:/usr/lib/systemd/system/avnav.service.d $ sudo nano avnav.conf
pi@Relexed:/usr/lib/systemd/system/avnav.service.d $ sudo systemctl daemon-reload
pi@Relexed:/usr/lib/systemd/system/avnav.service.d $ sudo systemctl enable avnav
pi@Relexed:/usr/lib/systemd/system/avnav.service.d $ sudo systemctl start avnav

Then I rebooted the system.

But now I don't have any chart anymore. And the o-charts button gives no information anymore (just a smily).

I hereby include the logfile.
[avnav-20241022.log](https://github.com/user-attachments/files/17482377/avnav-20241022.log)

Fietsenwiel, 2024-10-22:
In the logfile I see a permission denied. Is that the problem? I don't see it:

pi@Relexed:~ $ ls -al /home/pi/avnav/data/mapproxy/
totaal 40
drwxr-xr-x  6 avnav avnav 4096 21 okt 12:23 .
drwxrwxrwx 16 pi    pi    4096 22 okt 23:37 ..
-rw-r--r--  1 avnav avnav   96 24 mei 15:38 avnav_include.yaml
-rw-r--r--  1 avnav avnav 9958 21 okt 12:23 avnav_include.yaml.normal
drwxr-xr-x  3 avnav avnav 4096  2 okt 14:51 cache_data
drwxr-xr-x  2 avnav avnav 4096 24 mei 15:38 layers
drwxr-xr-x  2 avnav avnav 4096 24 mei 19:23 seed
drwxr-xr-x  2 avnav avnav 4096 24 mei 15:40 selections

free-x, 2024-10-23:
you don't see that the process belonging to the user “pi” cannot fully access the files belonging to the user “avnav”?

as user "pi" please execute following
```
cd
sudo chown -R pi.pi avnav
sudo systemctl restart avnav
```

Fietsenwiel, 2024-10-23:
Yes, thats it!

Thanks for your help!

closed 2024-10-23 by Fietsenwiel