wellenvogel/avnav #381: Trigger Redraw of External Widget no more working in master branch

kdschmidt1, 2024-11-03:
Trigger Redraw of External Widget results in "react-dom.development.js:4312 Uncaught TypeError: setUpdateCount is not a function" 
https://github.com/wellenvogel/avnav/blob/9fa9b5c37bd2a76527da828382a5988bc070110e/viewer/components/ExternalWidget.jsx#L65

quantenschaum, 2024-12-28:
Maybe 342245060a1fd4d23035c7cb2566ec1769145787 seems to break it again or something else is broken.

Map **overlay** widgets get only drawn once but are not updated.

Does the widget code need to be updated?

quantenschaum, 2024-12-29:
git bisect that 5934c79a0fb511413450cd27465574440bb3c798 is the breaking change.

wellenvogel, 2024-12-30:
Can you describe a bit more in detail what went wrong? Is triggerRedraw not working? Or is there no redraw on a store item change?

quantenschaum, 2024-12-30:
Uhm? It was working before 5934c79a0fb511413450cd27465574440bb3c798. The SailInstrument chart overlay is only drawn once and then it does not get updated. If the widget is in the widget areas it works fine. 

`renderCanvas` is called regularly, but it does not update on the screen.

wellenvogel, 2025-01-05:
I cannot reproduce your problem. I have a simple test widget that draws a rotating cross and that is working without any problems.
To support you I would need some code that I can use for testing.

quantenschaum, 2025-01-06:
Interesting. You can try the https://github.com/quantenschaum/Sail_Instrument plugin. Add it as a map overlay. The widget ist updated, but the overlay is not.

quantenschaum, 2025-01-06:
@kdschmidt1 do you have this problem, too?

quantenschaum, 2025-01-06:
Maybe something went wrong during merging on my side, I'll check that.

kdschmidt1, 2025-01-06:
I can confirm the error with the actual avnav master-branch and the original sailinstrument-plugin https://github.com/kdschmidt1/Sail_Instrument/releases/tag/20240731.
retesting with the latest avnav release branch https://github.com/wellenvogel/avnav/tree/release-20240616 is running properly

quantenschaum, 2025-01-06:
5934c79a0fb511413450cd27465574440bb3c798 causes the problem, but I do not understand why.

wellenvogel, 2025-01-07:
I tested with your proposed SI release and the SailInstrumentOverlay. But this draws nothing except the compass ring as it has no HDT available (although headingTrue is there). At the end there is no sail_instrument data at all - but the log gets filled with errors:
user-Sail_Instrument-ERROR-user-Sail_Instrument:setting invalid path gps.sail_instrument.HDT
So I don't think that there is any issue at the widget side.

wellenvogel, 2025-01-07:
Ok, seems to be a problem on my test setup with multiple instances...

wellenvogel, 2025-01-07:
Ok, after correcting my test setup I can reproduce the problem. Sometimes react18 has some strange effects...

quantenschaum, 2025-01-08:
It's working again!

wellenvogel, 2025-07-23:
release with [20250723](https://www.wellenvogel.de/software/avnav/downloads/release/20250723/?lang=en)

closed 2025-07-23 by wellenvogel