wellenvogel/avnav #383: oexserverd: spamming log files in bookworm 32bit

W-Geronius, 2024-11-03:
Using bookworm armhf-lite image, dmesg reveals hundreds of warnings like:
```
[  512.270341] usb 1-1.4: usbfs: process 3439 (oexserverd) did not claim interface 0 before use
[  512.360389] usb 1-1.4: usbfs: process 3447 (oexserverd) did not claim interface 0 before use
[  512.480331] usb 1-1.4: usbfs: process 3447 (oexserverd) did not claim interface 0 before use
[  518.320766] usb 1-1.4: usbfs: process 3461 (oexserverd) did not claim interface 0 before use
[  518.440577] usb 1-1.4: usbfs: process 3461 (oexserverd) did not claim interface 0 before use
[  518.550635] usb 1-1.4: usbfs: process 3471 (oexserverd) did not claim interface 0 before use
[  518.600569] usb 1-1.4: usbfs: process 3471 (oexserverd) did not claim interface 0 before use
[  599.891591] usb 1-1.4: usbfs: process 3590 (oexserverd) did not claim interface 0 before use
[  600.031197] usb 1-1.4: usbfs: process 3590 (oexserverd) did not claim interface 0 before use
[  600.141230] usb 1-1.4: usbfs: process 3598 (oexserverd) did not claim interface 0 before use
```
oesenc maps however get properly processed.
These messages are not present in arm64-lite image!

wellenvogel, 2024-11-03:
ocharts or ochartsng?
Which version?

W-Geronius, 2024-11-03:
ochartsng 20240616-raspbian-bookworm
the error persisted for 3 reboots, I checked dmesg because I was experiencing many wifi dropouts.
Now after 4th reboot, the messages are gone and WiFi is stable.
Magic.

I may reopen this issue as soon as I can safely reproduce.

closed 2024-11-03 by W-Geronius