wellenvogel/avnav #384: improved night mode

quantenschaum, 2024-11-12:
Improve the display of charts in night mode by applying CSS filters to the map canvas. 

This can be achieved in the `user.js` by

```js
const injectCSS = css => {
  let el = document.createElement('style');
  el.type = 'text/css';
  el.innerText = css;
  document.head.appendChild(el);
  return el;
};

injectCSS('.nightMode .ol-layer { filter: invert(100%) hue-rotate(180deg); }');
```

The brightness ist inverted and the hue rotating keep the colors correct.

quantenschaum, 2024-11-12:
I was not aware that this exists, that's even easier. It would make sense to disable this feature for charts that already come in a night theme.

wellenvogel, 2025-03-24:
I added a css class for the chart provider and a class for the chart name to the map object.
So the classes would look like 
` map int OSMOpenCPN2Balticmbtiles` for an internal mbtiles chart
` map systemochartsng GermanWaters20242024118` for a chart from ochartsng
` map systemmapproxy mpbsh` for a mapproxy chart

So this would allow to write a css rule like
```
.nightMode .map.int .ol-layer {
  filter: invert(100%) hue-rotate(180deg);
}
```
To address all internal charts (but not affecting charts from ocharts or mapproxy).

wellenvogel, 2025-07-23:
release with [20250723](https://www.wellenvogel.de/software/avnav/downloads/release/20250723/?lang=en)