wellenvogel/avnav #385: Connection errors in logfile

Fietsenwiel, 2024-11-14:
I see connection errors in the logfile but i don't make any connection. In Signal-k i disabled all connections:

`During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/usr/lib/avnav/server/handler/socketbase.py", line 111, in readSocket
raise Exception("no data received within timeout of %s seconds"%str(timeout))
Exception: no data received within timeout of 50.0 seconds

2024-11-14 13:28:55,362-592-905-signalk-INFO-disconnected from socket unknown connection
2024-11-14 13:28:57,364-592-905-signalk-INFO-STATUS[main] RUNNING, connected to localhost:10110
2024-11-14 13:28:57,364-592-905-signalk-INFO-successfully connected to localhost:10110
2024-11-14 13:28:57,365-592-905-signalk-INFO-STATUS[main] NMEA, connected to localhost:10110
2024-11-14 13:28:57,365-592-905-signalk-INFO-unknown connection established, start reading
2024-11-14 13:28:57,365-592-905-signalk-INFO-STATUS[main:#:reader] RUNNING, receiving unknown connection
2024-11-14 13:29:47,447-592-905-signalk-ERROR-exception while reading from socket: Traceback (most recent call last):
File "/usr/lib/avnav/server/handler/socketbase.py", line 102, in readSocket
data = sock.recv(1024)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/usr/lib/avnav/server/handler/socketbase.py", line 111, in readSocket
raise Exception("no data received within timeout of %s seconds"%str(timeout))
Exception: no data received within timeout of 50.0 seconds

2024-11-14 13:29:47,448-592-905-signalk-INFO-disconnected from socket unknown connection
2024-11-14 13:29:49,449-592-905-signalk-INFO-STATUS[main] RUNNING, connected to localhost:10110
2024-11-14 13:29:49,449-592-905-signalk-INFO-successfully connected to localhost:10110
2024-11-14 13:29:49,450-592-905-signalk-INFO-STATUS[main] NMEA, connected to localhost:10110
2024-11-14 13:29:49,450-592-905-signalk-INFO-unknown connection established, start reading
2024-11-14 13:29:49,450-592-905-signalk-INFO-STATUS[main:#:reader] RUNNING, receiving unknown connection`

Everything works fine but this should not happen gues?
Only somtimes, the o-charts disapears, restarting avnav brings it back then.

The Avnav version is 20240616 on a Raspberry pi 4b.

wellenvogel, 2024-11-14:
Would be helpful to attach your config...
But from the log:
Normally socket connections in AvNav will close and reopen if there is no data received within the timeout (otherwise you would potentially not detect a dead connection).
And this messages are created from a SocketReader you named "signalk". If you don't route data through SignalK you most probably don't need this reader. So just disable it.
The connection settings in SignalK are not relevant for this issue.

> Only somtimes, the o-charts disapears, restarting avnav brings it back then.

Not sure what exactly this means. Especially: Which ocharts version (ochartsng?), what means "disapears", what do you restart...
Btw: please create separate issue for different problems.

Fietsenwiel, 2024-11-14:
Here is my config:

`<?xml version="1.0" ?><AVNServer>

<AVNConfig debug="0" debugToLog="true" expiryTime="300" logfile="/home/pi/avnav/data/log/avnav.log" maxtimeback="5" ownMMSI="244198117" settimecmd="" settimeperiod="3600" systimediff="5"/>

<Workers>
<AVNGpsdFeeder gpsdcommand="/usr/sbin/gpsd -b -n -N -G" listenerPort="4020" port="2948" useGpsd="false"/>

<AVNHttpServer httpPort="8080" index="/viewer/avnav_viewer.html" navurl="/viewer/avnav_navi.php">
	<Directory path="/home/pi/avnav/data/charts" urlpath="maps"/>
	<Directory path="/home/pi/avnav/data/log" urlpath="log"/>
	<Directory path="/home/pi/avnav/data/tracks" urlpath="tracks"/>
	<Directory path="/home/pi/avnav/data/routes" urlpath="routes"/>
	<MimeType extension=".avt" type="text/plain"/>
	<MimeType extension=".log" type="text/plain"/>
</AVNHttpServer>

<AVNTrackWriter interval="10" mindistance="50" trackdir="/home/pi/avnav/data/tracks"/>

<AVNSocketWriter blackList="signalk" filter="" maxDevices="5" minTime="50" name="tosignalk" port="28628" read="true"/>

<AVNSocketReader filter="^$RMB" host="localhost" name="signalk" port="10110"/>
<AVNPluginHandler>
        <builtin-signalk enabled="true" migrated="True"/>

<system-history period="30.0" sensorNames="" storeKeys="gps.speed,gps.signalk.navigation.attitude.roll,gps.signalk.environment.inside.temperature,gps.signalk.environment.inside.pressure,gps.transducers.HumidityBinnen,gps.signalk.temp-buiten-temperature,gps.signalk.temp-koelkast-temperature,gps.signalk.temp-binnen-temperature" storeTime="72"/>
<builtin-canboat enabled="False"/>
<user-seatalk-remote enabled="False"/>
<builtin-Sail_Instrument TWD_filtFreq="0.2"/>
<system-Sail_Instrument allow_fallback="True" calc_vmc="False" ground_wind="" gybe_angle="0" laylines_from_matrix="False" lee_factor="10" minmax_samples="200" show_polar="True" smoothing_factor="0.1" tack_angle="0" write_data="False" period="1" laylines_polar="False" wmm_file="WMM2020.COF" wmm_period="600" depth_transducer="-1" draught="-1" nmea_write="False" nmea_filter="" nmea_priority="10" nmea_id="CA" nmea_decode="True"/>
</AVNPluginHandler>

<AVNChartHandler/>

<AVNRouter routesdir="/home/pi/avnav/data/routes" useRhumbLine="True"/>

<AVNNmeaLogger/>
<AVNImporter importDir="/home/pi/avnav/data/import"/>

<AVNCommandHandler>
	<Command command="sudo shutdown -P" name="shutdown"/>
	<Command command="/usr/lib/python3/dist-packages/openplotterAvnav/data/sound.sh 90%" name="sound" repeat="1"/>
        <Command command="/usr/bin/python" name="pythonalarm"/>
        <Command command="collision" name="sound" repeat="1"/>
</AVNCommandHandler>

<AVNAlarmHandler>
        <Alarm category="info" name="waypoint" repeat="1"/>
        <Alarm category="info" name="connectionLost" repeat="0"/>
        <Alarm category="critical" name="anchor" repeat="20000"/>
        <Alarm category="critical" name="gps" repeat="20000"/>
        <Alarm category="critical" name="mob" repeat="2"/>
        
        <Alarm category="critical" command="sinkingAlarm" name="sk:sinking" repeat="2"/>
        <Alarm category="critical" name="sk:danger.collision" repeat="1"/>
        <Alarm command="pythonalarm" name="anchor" parameter="/home/pi/scripts/anchoralarm.py" repeat="1"/>
</AVNAlarmHandler>
  
</Workers>
<AVNUserAppHandler><UserTool icon="/user/images/SPDicoon.PNG" keepUrl="True" name="854e069889c4ec91d2004ab3f5ab1937" newWindow="false" title="Sailing Performance Dashboard" url="http://$HOST/SPD/spd.html"/>
<UserTool icon="/user/images/radio-transmitter.png" keepUrl="True" name="e189f2009cea2f22ba826a801bf03c5f" newWindow="false" title="None" url="http://$HOST:3000/vhfinfo"/>
<UserTool icon="/user/images/Metertje.PNG" keepUrl="True" name="a4fe4205ec9c435b8741f64a3564edfc" title="" url="http://$HOST:3000/@mxtommy/kip/#/page/0"/>
<UserTool icon="/user/images/Zeilboot.png" keepUrl="True" name="673f80de3873186cdb9732df5464760e" newWindow="false" url="http://$HOST:3000/signalk-trim-plugin/#"/>
<UserTool icon="/user/images/Windvaan.PNG" keepUrl="True" name="12b16a34b7814904e512e8552690df5f" newWindow="true" url="https://nl.windfinder.com/#10/51.7785/4.5607"/>
<UserTool icon="/user/images/Windvaan.PNG" keepUrl="True" name="5cb4274b2967130112a04c51636f1db8" title="" url="http://weerlive.nl/index.php?pagina=hwind"/>
<UserTool icon="/user/images/Docs.png" keepUrl="True" name="3515f96cb20596a49b16c8f7a4f1a867" title="" url="http://$HOST:80/Relexed.html"/>
<UserTool icon="/user/images/log-icon.png" keepUrl="True" name="c8cb7a6bb0849c9fa616d47684ac69fe" newWindow="true" title="elog" url="http://$HOST:9091/relexed/"/>
<UserTool icon="/user/images/S100.png" keepUrl="True" name="8bbf1233b57e02156658b05b82d095e0" newWindow="false" title="null" url="http://$HOST:3000/@signalk/signalk-autopilot/"/>
<UserTool icon="/user/images/Tiden.png" keepUrl="True" name="f0082465a6ba9b4e8ef63d281de243a2" newWindow="false" url="http://$HOST:3000/sk-tidal-dashboard"/>
<UserTool icon="/user/images/zigbee2mqtt.jpg" url="http://$HOST:8090/#/dashboard" newWindow="true" keepUrl="True" name="a91922dfeea560a846605da9ca47bc9b"/>
</AVNUserAppHandler>

<AVNSignalKHandler fetchCharts="False" migrated="True" password="alsk1298" receiveNotifications="True" sendData="True" userName="pi" uuid="23e61eee-cf4e-461c-8b6c-e283d627731f"/>
</AVNServer>`

I do use signalk but I disabled all connections to see if the problem would disappear. It didn't.
And I thought that disapearing of o-charts (no ochart maps available anymore) was related to this problem. If not I will make an other issue.

wellenvogel, 2024-11-14:
So as I wrote...
If there is no data on the signalk connection it will close and reopen.
If you do not want this, set the timeout at this connection to a very large value like 31104000...
For sure not related to any ocharts issue.
But:

> no ochart maps available anymore

is not very clear...
If this happens on the main page - we need all the logs (avnav, ocharts) and all version + system info.

Fietsenwiel, 2024-11-16:
I only see this in the logfile. Everything seems to work fine. So, is it actualy a problem?

It's on a rapsberry pi 4b
Linux Relexed 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Versions:
[image]
And the logfile:
[avnav-20241116.log](https://github.com/user-attachments/files/17784569/avnav-20241116.log)

And here is a part of the o-charts log (from wher its goes wrong):

2024/11/14-15:57:05.084-0x7f4fffec00-INFO-CacheFiller CS_home_pi_avnav_data_ocharts_charts_oeuSENC-BE_NL_UK-2022-2-0-base-sgl001DF950: prefilled for zoom=13: 405
2024/11/14-15:57:05.086-0x7f4fffec00-INFO-CacheFiller CS_home_pi_avnav_data_ocharts_charts_oeuSENC-BE_NL_UK-2022-2-0-base-sgl001DF950: prefilled for zoom=12: 257
2024/11/14-15:57:05.086-0x7f4fffec00-INFO-Cache filler CS_home_pi_avnav_data_ocharts_charts_oeuSENC-BE_NL_UK-2022-2-0-base-sgl001DF950 prefilled 70296 tiles
2024/11/14-15:57:05.086-0x7f4fffec00-INFO-Cache filler finished
2024/11/16-07:01:52.055-0x7f7311fc00-ERROR-read(fd=12): SYSERROR: 104=Connection reset by peer
2024/11/16-07:01:52.058-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 32=Broken pipe
2024/11/16-07:01:52.058-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 0=Success
2024/11/16-07:01:52.058-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 32=Broken pipe
2024/11/16-07:01:52.059-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 0=Success
2024/11/16-07:01:52.061-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 32=Broken pipe
2024/11/16-07:01:52.061-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 0=Success
2024/11/16-07:01:52.062-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 32=Broken pipe
2024/11/16-07:01:52.062-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 0=Success
2024/11/16-07:01:52.064-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 32=Broken pipe
2024/11/16-07:01:52.064-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 0=Success
2024/11/16-07:01:52.065-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 32=Broken pipe
2024/11/16-07:01:52.065-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 0=Success
2024/11/16-07:01:52.067-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 32=Broken pipe
2024/11/16-07:01:52.067-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 0=Success
2024/11/16-07:01:52.067-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 32=Broken pipe
2024/11/16-07:01:52.068-0x7f7311fc00-ERROR-write(fd=12): SYSERROR: 0=Success
2024/11/16-07:01:52.077-0x7f7311fc00-ERROR-GetLocalAddress(fd=12): SYSERROR: 107=Transport endpoint is not connected
2024/11/16-07:01:52.080-0x7f7311fc00-ERROR-write(fd=17): SYSERROR: 32=Broken pipe
2024/11/16-07:01:52.080-0x7f7311fc00-ERROR-write(fd=17): SYSERROR: 0=Success
2024/11/16-07:01:52.081-0x7f7311fc00-ERROR-write(fd=17): SYSERROR: 32=Broken pipe
2024/11/16-07:01:52.081-0x7f7311fc00-ERROR-write(fd=17): SYSERROR: 0=Success

wellenvogel, 2024-11-16:
could we go for an own issue with ocharts?
The broken pipes are pointing to a networking issue (the browser was running into a timeout and closing the connection).

Fietsenwiel, 2024-11-17:
A made another issue for o-charts disapearing.

I guess the problem about connection errors in Signalk in the logfile is nothing to be worried about...

wellenvogel, 2024-11-28:
closing this as it is no real problem