wellenvogel/avnav #386: Ocharts disappear after a while

Fietsenwiel, 2024-11-16:
My O-charts disappear after a while. I don't know after how long but that could be after one day.

After restarting AvNav the O-charts are back, again some time.

I made pictures of the situation and I enclose the logfile.
This is on a Raspberry pi 4b.

[avnav-20241116.log](https://github.com/user-attachments/files/17784554/avnav-20241116.log)
[image]
[image]
[image]
[image]
[image]

Fietsenwiel, 2024-11-17:
Here the issue with O-charts disapearing.

You say: "The broken pipes are pointing to a networking issue (the browser was running into a timeout and closing the connection)."

I have three wifi connections:

[image]

Wlan0 connects to an accespoint
Wlan1 is an accespoint to acces the raspberry directly when sailing.
tun0 is a vpn tunnel over wlan1 to acces the raspberry pi from home and to get info about humidity, temp etc.

I know tha o-charts don't work over the vpn.

All three connections work without problems (for as I know).

Can you explain how network issues make o-charts diaspear but everything else keeps working?
This might help solving this.

Fietsenwiel, 2024-11-17:
I have no idea where to look. Just looking around:
But I see is the /var/log/syslog "TCP: request_sock_TCP: Possible SYN flooding on port 8080. Sending cookies.  Check SNMP counters."
Is this a problem?

free-x, 2024-11-17:
very interesting
What is output of
```
ss -nta | grep 8080
```

Fietsenwiel, 2024-11-17:
ss -nta | grep 8080

LISTEN     0      5                     0.0.0.0:8080                0.0.0.0:*
ESTAB      0      0               192.168.0.198:8080           192.168.0.85:63677
ESTAB      0      0               192.168.0.198:8080           192.168.0.85:63683
ESTAB      0      0               192.168.0.198:8080           192.168.0.21:54418
ESTAB      0      0               192.168.0.198:8080           192.168.0.21:51687
ESTAB      0      0               192.168.0.198:8080           192.168.0.85:63692
ESTAB      0      0               192.168.0.198:8080           192.168.0.85:63458

But I think I have found the cause of the network problems. The ss -nta | grep 8080 output is after "solving" the problem.
Deactivation the VHFInfo Signalk plugin solved the problem.
The VHFinfo helps finding the right VHF channel at the right places.

Somehow the plugin does something wrong when there is no GPS signal, its keeps complaining about it and it looks like it leads to signalk timeouts. A lot of them:

info is missing (navigation.position)
Nov 17 00:02:41 Relexed signalk-server[28812]: 2024-11-16T23:02:41.364Z vhfinfo searchPolygon is null
Nov 17 00:02:46 Relexed signalk-server[28812]: 2024-11-16T23:02:46.365Z vhfinfo createSearchPolygon: coordinates info is missing (navigation.position)
Nov 17 00:02:46 Relexed signalk-server[28812]: 2024-11-16T23:02:46.366Z vhfinfo searchPolygon is null
Nov 17 00:02:51 Relexed signalk-server[28812]: 2024-11-16T23:02:51.366Z vhfinfo createSearchPolygon: coordinates info is missing (navigation.position)

But I still don't understand the relation between o-charts and network issues.

Fietsenwiel, 2024-11-18:
I have made an issue at Github of o-charts:
[https://github.com/htool/vhfinfo/issues/16](url)

But there tey say that the problem is the o-charts plugin

wellenvogel, 2024-11-18:
Are you sure about the link?

Fietsenwiel, 2024-11-22:
Ok, thanks. But i have given up. I just stop using VHFinfo.

wellenvogel, 2024-11-22:
Ok, I see. Would have been good to find the problem...

closed 2024-11-22 by wellenvogel