wellenvogel/avnav #387: Error in pixelToLonLat

kdschmidt1, 2024-12-03:
function pixelToLonLat of mapWidgetContext is running into Error:
 "TypeError: _this2.drawing.pointFromMap is not a function"
(actual master branch and also release 20240616)

wellenvogel, 2024-12-04:
Can you describe this in more detail please? I.e. in which code do you call pixelToLonLat?

kdschmidt1, 2024-12-04:
Hello Andreas,
I’m currently working on a Grib-Plugin.
For server-requests i need to know the currently visible screen-extent in lat/lon units.
So i tried to calculate the edges by calling self.lonLatToPixel(0,0) and self.lonLatToPixel(canvas.width, canvas.height) inside the renderCanvas-function of the map-Plugin, but this ran into the mentioned error.

wellenvogel, 2024-12-04:
Sounds cool...
Do you have the js code online- or can you send it to me?
I tried - but cannot reproduce the problem.

kdschmidt1, 2024-12-04:
Hi Andreas,
currently the Grib-Plugin repository is not published. I dont like to share the code already now with you (it is "spaghetti"), but i can reproduce the error also in the current release of the  [Sailinstrument Plugin](https://github.com/kdschmidt1/Sail_Instrument/releases/tag/20240503) adding `test=this.pixelToLonLat(0,0)` in line 274 after `renderCanvas: function(canvas, `data){`
(I will send you a modified js-file by mail)
You can easily check also by setting a breakpoint at this line and try to execute the code in the console resulting in the mentioned error (at least on my system) with the todays master branch of your repository.
my development system is ubuntu 22.04 with python 3.10.12, using Chrome Version 130.0.6723.91
for the grib plugin i will anyway come back to you asap (hopefully end of next week) with a more structured code and some documentation to know your opinion if it is worth to go on, it might need some additional effort also on avnav (especially for the GUI) if one likes to use it with touchscreen! What i can mention already now is that the map plugins are hiding the boat symbol, there seems to be an error with the organization of the map-layers.

kdschmidt1, 2024-12-05:
mabe i shoud mention, that the error is not shown in the console, because it is caught in the try/catch block in mapwidget.jsx.
see my screenshots attached.
[image]
[image]

wellenvogel, 2024-12-05:
could reproduce + fix it...