wellenvogel/avnav #389: Question: slow updating sensors

Fietsenwiel, 2024-12-10:
I have some Zigbee sensors that shows their values in Signalk.
And in Avnav I have some gauge to show heir value.

But zigbee devices somtimes update afte 20 minutes or longer.
It look like Avnav does not like that en often show no value on the gauge:

[image]

Also in the grafics you see interruped values in the humidity graph:

[image]

Zigbee always shows a value:
[image]

Is there a way to keep the last value in Avnav for a much longer time?

wellenvogel, 2024-12-10:
expiryTime in AVNConfig.
Global for all received values.

Fietsenwiel, 2024-12-10:
Ah, so simple, i'll try that!
Thanks!

Fietsenwiel, 2024-12-10:
Tested and it works!

closed 2024-12-10 by Fietsenwiel