wellenvogel/avnav #392: cannot edit settings

quantenschaum, 2024-12-30:
On current master: when trying to change the settings (navigation, ais, ...) I can change a value, but when clicking OK in the dialog, the dialog closes and the value is back at its old value.

Also, when changing the config of widgets, I often have to make a change twice to get it actually applied (in selections lists). 

Changing the configuration of plugins does work correctly.

quantenschaum, 2024-12-31:
This seems to affect only settings where a value has to be entered (numbers). Selecting from a list of options does work.

wellenvogel, 2024-12-31:
Just rewrote all the dialog stuff - so there still could be errors from all those changes...
Many thanks for all the reports.

quantenschaum, 2025-01-01:
bisection yields that the problem was introduced with ec0e77ed1659d930b178b4d89240481156c1daf1

wellenvogel, 2025-07-23:
release with [20250723](https://www.wellenvogel.de/software/avnav/downloads/release/20250723/?lang=en)

closed 2025-07-23 by wellenvogel