wellenvogel/avnav #393: AIS infopage crashes

quantenschaum, 2024-12-30:
I created a simple plugin https://github.com/quantenschaum/aisstream-plugin to pull AIS data from https://aisstream.io/ into AVnav.

The plugin injects the AIS data as NMEA sentences.

I works, the AIS targets show up on the map. But when I click on a target, the AIS infopage opens, displays the data shortly, then all data fields become empty and the page closes, the map is shown again and all AIS target are gone, but reappear quickly.

quantenschaum, 2024-12-31:
It does not happen on all AIS targets, for some the info page is displayed as expected, but for many others it crashes as described.

quantenschaum, 2024-12-31:
Here is some test data (click on the vessels in the Kadetrinne)

```
!AIVDM,2,1,0,A,538;tn02:aWkD8hsD01<d4pT400000000000000t2rG7A0000?4mjCcFDm0A,0*06
!AIVDM,2,2,0,A,;DmjCP00000,2*4F
!AIVDM,2,1,0,A,539Np7000000@d`o801LTp@<5B3;P0000000000`1p32400004lSlm3hjw?`,0*0D
!AIVDM,2,2,0,A,0PC52@p<P00,2*6C
!AIVDM,1,1,,A,33lTF05000PoPD6NwEeWCq0F2000,0*78
!AIVDM,1,1,,A,13cJ>B0uAJPolPtO;eiGVV:J0000,0*07
!AIVDM,1,1,,A,13AIvT0ti80p1HdO=FN7uVNJ0000,0*2C
!AIVDM,1,1,,A,13:0DP0P00Pr5ujO9Epv4?vR2000,0*4A
!AIVDM,1,1,,A,139uOk0P0tPr2>VO952qOwvVR000,0*6A
!AIVDM,1,1,,A,139KUE0000PoNVNNwveMuTbb0000,0*69
!AIVDM,2,1,0,A,539EIqP2>PN<@8QT000LtE8lUA`000000000000o0hN3600004TSlm3hjh00,0*2E
!AIVDM,2,2,0,A,00000000000,2*24
!AIVDM,1,1,,A,139UO55P1>PoLTPNwJarNgvp0000,0*60
!AIVDM,2,1,0,A,5815RMP2DiMAKL7GOS44AHDqAE8D000000000016C8KC;0000AlSlm3hjh00,0*06
!AIVDM,2,2,0,A,00000000000,2*24
!AIVDM,2,1,0,A,539wa@P00000@dLk800LDv1=A84i=Dp@0000001J1H7230000311DTjp4mDU,0*7D
!AIVDM,2,2,0,A,QFH11DTjh00,2*59
!AIVDM,2,1,0,A,53?vj`000000@487800LDv0M850P00000000001S1@8330000311DTjsDmDU,0*54
!AIVDM,2,2,0,A,QFKA1DTjh00,2*2A
!AIVDM,1,1,,A,3815RMU000PoP=RNw6s0AE5L0000,0*62
!AIVDM,1,1,,A,13ku;AD01`0oP2jO9kHbLpCH0000,0*32
!AIVDM,1,1,,A,13thpD002iPquFdOSPK6f5T60000,0*21
!AIVDM,1,1,,A,139Ilr0P0hPqb0LO:0q:nwwd2000,0*4B
!AIVDM,1,1,,A,13aLmW@000PoJsrNw=CS1m240000,0*79
!AIVDM,1,1,,A,E>j9dT@12h7aah;TW230a6P0000@Ol2W?ckFp00000NP0000000000000000,4*69
!AIVDM,1,1,,A,E>j9dp0;TW2@30a6h10V:4QpW;P@Lr65?WwNH00000MP2000000000000000,4*57
!AIVDM,2,1,0,A,539UO5000000@K?S?<1<DhDpD0000000000000150hD4300000P000000000,0*4F
!AIVDM,2,2,0,A,00000000000,2*24
!AIVDM,1,1,,A,E>j9e42;Uh7aah860b37a6P0000@P=Wm?d3?p00000NP0000000000000000,4*71
!AIVDM,1,1,,A,Evj9e10;Uh7Fprh;TW230a6P000@P?2L?dj3h00000300000000000000000,4*0D
!AIVDM,2,1,0,A,53aIgt0000010CKGOT04PuV3G00000000000000Q2hS8A0000513lQ4Q@j50,0*2F
!AIVDM,2,2,0,A,00000000000,2*24
!AIVDM,1,1,,A,339EIqUP000oOMDNutk00?v<0000,0*2F
!AIVDM,2,1,0,A,542OaR@2CqCq@<7OC3MBpeE9D<E<lD000000001@IHaO;0000U4381i3P000,0*17
!AIVDM,2,2,0,A,00000000000,2*24
!AIVDM,1,1,,A,33cJ>B001KPol;lO;`4oTn4J0000,0*06
!AIVDM,1,1,,A,33thpD002iPquOhOSKnFeUTn0000,0*53
!AIVDM,1,1,,A,139KUE0000PoNV4Nwvd=uTbb0000,0*62
!AIVDM,1,1,,A,13AIvT0uA80p0rHO=AFWq6Lj0000,0*3B
!AIVDM,1,1,,A,139UO55P1<PoKOdNwJPbPwvp0000,0*65
!AIVDM,1,1,,A,33`lRoE000PoIm8Nw:@C`s4p0000,0*16
!AIVDM,1,1,,A,13cJ>B0uAKPol5<O;V2oRV540000,0*1E
!AIVDM,1,1,,A,339Np75P000oLAlNvrBu7gw60000,0*21
!AIVDM,1,1,,A,19NSI4042:PsKj<OVos3dk1:0000,0*4E
!AIVDM,1,1,,A,33?vj`50000oIfPO08fe3@ID0000,0*7A
!AIVDM,1,1,,A,339KUE0000PoNV:NwveMuTcB0000,0*3E
!AIVDM,1,1,,A,35?5@650000oOT4NwA<0u57F0000,0*6C
!AIVDM,2,1,0,A,53cJ>B02A?;i0<pd000L45=@8u8L000000000016;8;870000?A1DTjh0000,0*24
!AIVDM,2,2,0,A,00000000000,2*24
!AIVDM,1,1,,A,13aLmW@000PoJt0Nw=CS1m240000,0*3C
!AIVDM,1,1,,A,139uOk0P0tPr11RO91epo?wnR000,0*61
!AIVDM,1,1,,A,13ku;AD1ia0oN84O9jwbM`B40000,0*72
!AIVDM,1,1,,A,33aIgt5P00PoK;tNwD4P0?v>2000,0*1F
```

quantenschaum, 2024-12-31:
The JS console does not show any error messages.

wellenvogel, 2024-12-31:
Thanks for reporting - will have a look.

quantenschaum, 2025-01-01:
It happens not only on current master but also in version 20241128.

quantenschaum, 2025-01-01:
It looks like this.

https://github.com/user-attachments/assets/e3338fee-2cff-4deb-8ddc-ad451c7d2bb1

quantenschaum, 2025-01-02:
I found something: The "crash" happens when the target is outside the AIS-range (config value). Clicking on a target inside that range from the own boat works fine. When panning away from the boat, target inside the AIS range from the map center are displayed as well. When clicking on a target inside this range but outside the range from own boat, the page "crashes". I think, when clicking the target and the info page opens, the map center is lost, the target gets out of range and is no longer displayed.

quantenschaum, 2025-01-07:
That fixed it.

I have a question concerning AIS. I did not find anything in the docs. 
What does the field `we pass` and the values `done, front, back, pass, ...` actually mean?
I would guess
- done = CPA already passed, TCPA<0, but there are targets with TCPA<0 that show `front/back`
- front/back = we pass in front/back of the target or does the target pass in front/back of us?
- pass?
What are the rules for calculating these?

wellenvogel, 2025-01-08:
see #397

closed 2025-01-09 by quantenschaum