wellenvogel/avnav #401: Suggestion: adding settings to some widgets

gtf375, 2025-01-23:
Could you add a distance formatter to the Active Route widget so that you can choose to display the distance in meters or kilometers? The distance in nautical miles is not always convenient.
Also, reduce the minimum XTE value to 0.03 nm, or better yet, make the setting in meters, and the minimum value is 50 meters. On narrow channels and rivers 0.1 nm, this is too much deviation.

wellenvogel, 2025-01-30:
Using different units is for sure really interesting.
But when starting this I would like to go for some global settings to change the units.
In the mean time you could create an own widget/formatter to do what you want.
Should be fairly straight forward...
https://www.wellenvogel.net/software/avnav/docs/hints/userjs.html?lang=en#h2:Widgets

wellenvogel, 2025-07-23:
for XTE release with [20250723](https://www.wellenvogel.de/software/avnav/downloads/release/20250723/?lang=en)

wellenvogel, 2026-11-02:
implemented with [20261102](https://www.wellenvogel.de/software/avnav/newDoc/latest/en/installation/versions/20261102.html)

closed 2026-11-02 by wellenvogel