wellenvogel/avnav #404: After update: no AIS targets shown

ukoehler, 2025-02-23:
I updated the software on my RaspPi Openplotter today and AvNav does not show any AIS tragets anymore.

Signal K Server: version 2.13.2
AvNav: Version: 20240616

I used OpenCPN on my Linux Laptop Connected to the SignalK server to confirm, that my AIS setup is working.

AvNav Server Staus:
```
[11]SignalKHandler
￼
￼
alarms
sender ready
￼
authentication
successfully authenticated locally admin
￼
charts
disabled
￼
main
error when fetching from signalk http://localhost:3000/signalk/v1/api/
￼
source
avnav-bffdaf80-3694-4884-a3e0-ec22412be01a
￼
timeoffset
time offset=-0.40378403663635254
￼
websocket
connected at ws://localhost:3000/signalk/v1/stream, timeOffset=-0s
￼
write
connected at ws://localhost:3000/signalk/v1/stream, timeOffset=-0s
```

Trying the failing link above http://localhost:3000/signalk/v1/api/ (with the correct ip address) returns the required data?
```
{"vessels":{"urn:mrn:imo:mmsi:218050350":{"mmsi":"218050350","name":"CODDI","communication":{"callsignVhf":"DG9593"},"design":{"draft":{"meta":{"description":"The draft of the vessel","properties":{"minimum":{"description":"The minimum draft of the vessel","type":"number","units":"m"},"maximum":{"description":"The maximum draft of the vessel","type":"number","units":"m"},"current":{"description":"The current draft of the vessel","type":"number","units":"m"},"canoe":{"description":"The draft of the vessel without protrusions such as keel, centerboard, rudder","type":"number","units":"m"}}},"value":{"maximum":1.9},"$source":"defaults","timestamp":"2025-02-23T20:41:13.453Z"},"length":{"meta":{"description":"The various lengths of the vessel","properties":{"overall":{"type":"number","description":"Length overall","units":"m"},"hull":{"type":"number","description":"Length of hull","units":"m"},"waterline":{"type":"number","description":"Length at waterline","units":"m"}}},"value":{"overall":13},"$source":"can0.43","timestamp":"2025-02-23T22:12:37.730Z","values":{"defaults":{"value":{"overall":13.41},"timestamp":"2025-02-23T20:41:13.453Z"},"can0.43":{"value":{"overall":13},"timestamp":"2025-02-23T22:12:37.730Z","pgn":129810}},"pgn":129810},"beam":{"meta":{"units":"m","description":"Data should be of type number."},"value":4,"$source":"can0.43","timestamp":"2025-02-23T22:12:37.730Z","values":{"defaults":{"value":4.18,"timestamp":"2025-02-23T20:41:13.453Z"},"can0.43":{"value":4,"timestamp":"2025-02-23T22:12:37.730Z","pgn":129810}},"pgn":129810},"aisShipType":{"meta":{"description":"The ais ship type see http://www.bosunsmate.org/ais/message5.php","properties":{"id":{"type":"number","description":"The ship type number"},"name":{"type":"string","description":"The ship type name"}}},"value":{"id":36,"name":"Sailing"},"$source":"can0.43","timestamp":"2025-02-23T22:12:37.730Z","values":{"defaults":{"value":{"name":"Sailing","id":36},"timestamp":"2025-02-23T20:41:13.453Z"},"can0.43":{"value":{"id":36,"name":"Sailing"},"timestamp":"2025-02-23T22:12:37.730Z","pgn":129810}},"pgn":129810}},"navigation":{"position":{"meta":{"description":"The position of the vessel in 2 or 3 dimensions (WGS84 datum)","properties":{"longitude":{"type":"number","description":"Longitude","units":"deg","example":4.98765245},"latitude":{"type":"number","description":"Latitude","units":"deg","example":52.0987654},"altitude":{"type":"number","description":"Altitude","units":"m"}}},"value":{"longitude":-60.8811266,"latitude":14.4611066},"$source":"can0.43","timestamp":"2025-02-23T22:18:17.136Z","pgn":129025},"speedOverGround":{"meta":{"units":"m/s","description":"Vessel speed over ground. If converting from AIS 'HIGH' value, set to 102.2 (Ais max value) and add warning in notifications"},"value":0.04,"$source":"can0.43","timestamp":"2025-02-23T22:18:17.086Z","pgn":129026},"datetime":{"meta":{"description":"Time and Date from the GNSS Positioning System"},"value":"2025-02-23T22:18:17Z","$source":"can0.43","timestamp":"2025-02-23T22:18:17.107Z","pgn":129029},"gnss":{"antennaAltitude":{"meta":{"units":"m","description":"Altitude of antenna"},"value":9.5,"$source":"can0.43","timestamp":"2025-02-23T22:18:17.107Z","pgn":129029},"geoidalSeparation":{"meta":{"description":"Difference between WGS84 earth ellipsoid and mean sea level"},"value":-42.69,"$source":"can0.43","timestamp":"2025-02-23T22:18:17.107Z","pgn":129029},"methodQuality":{"meta":{"description":"Quality of the satellite fix"},"value":"GNSS Fix","$source":"can0.43","timestamp":"2025-02-23T22:18:17.107Z","pgn":129029},"integrity":{"meta":{"description":"Integrity of the satellite fix"},"value":"no Integrity checking","$source":"can0.43","timestamp":"2025-02-23T22:18:17.107Z","pgn":129029}},"courseRhumbline":{"nextPoint":{"position":{"meta":{},"value":{"longitude":-60.86833333333333,"latitude":14.468},"$source":"fromAvnav.GP","timestamp":"2025-02-23T22:18:15.906Z","values":{"avnav-bffdaf80-3694-4884-a3e0-ec22412be01a":{"value":null,"timestamp":"2025-02-23T20:41:31.016Z"},"fromAvnav.GP":{"value":{"longitude":-60.86833333333333,"latitude":14.468},"timestamp":"2025-02-
.....
```
I attched the complete output as a file.

[SignalKResponse.txt](https://github.com/user-attachments/files/18933852/SignalKResponse.txt)

Here is the end of the log file with an error message that might help:

```
File "/usr/lib/avnav/server/handler/signalkhandler.py", line 1635, in iterateToValue
callback(prefix,node.get('value'),node.get('$source'),node.get('timestamp'))
File "/usr/lib/avnav/server/handler/signalkhandler.py", line 1656, in store
if self.checkOutdated(timestamp):
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/avnav/server/handler/signalkhandler.py", line 1243, in checkOutdated
if timeStamp < oldest:
^^^^^^^^^^^^^^^^^^
TypeError: '<' not supported between instances of 'NoneType' and 'float'

2025-02-23 22:19:20,241-9520-9538-SignalKHandler-INFO-STATUS[main] ERROR, error when fetching from signalk http://localhost:3000/signalk/v1/api/
2025-02-23 22:19:21,171-9520-9542-SKSendAlarms-INFO-STATUS[alarms] NMEA, sender ready
2025-02-23 22:19:25,242-9520-11515-signalk-websocket-websocket-INFO-websocket connection websocket closed
2025-02-23 22:19:25,243-9520-11515-signalk-websocket-websocket-INFO-STATUS[websocket] ERROR, connection closed at ws://localhost:3000/signalk/v1/stream
2025-02-23 22:19:25,245-9520-11515-signalk-websocket-websocket-INFO-websocket receiver websocket finished
2025-02-23 22:19:25,259-9520-9538-SignalKHandler-INFO-found api url http://localhost:3000/signalk/v1/api/
2025-02-23 22:19:25,260-9520-9538-SignalKHandler-INFO-using websockets at ws://localhost:3000/signalk/v1/stream, querying with period 30
2025-02-23 22:19:25,260-9520-9538-SignalKHandler-INFO-STATUS[websocket] STARTED, connecting at ws://localhost:3000/signalk/v1/stream
2025-02-23 22:19:25,261-9520-9538-SignalKHandler-INFO-websocket websocket created at ws://localhost:3000/signalk/v1/stream
2025-02-23 22:19:25,262-9520-11522-signalk-websocket-websocket-INFO-websocket receiver websocket started
2025-02-23 22:19:25,264-9520-9538-SignalKHandler-INFO-STATUS[write] STARTED, connecting at ws://localhost:3000/signalk/v1/stream
[... 5 lines cut ...]
File "/usr/lib/avnav/server/handler/signalkhandler.py", line 1189, in _runI
self.storeData(data,self.config.priority)
File "/usr/lib/avnav/server/handler/signalkhandler.py", line 1660, in storeData
self.iterateToValue(node,None,store)
File "/usr/lib/avnav/server/handler/signalkhandler.py", line 1640, in iterateToValue
self.iterateToValue(v,path,callback)
File "/usr/lib/avnav/server/handler/signalkhandler.py", line 1640, in iterateToValue
self.iterateToValue(v,path,callback)
File "/usr/lib/avnav/server/handler/signalkhandler.py", line 1640, in iterateToValue
self.iterateToValue(v,path,callback)
File "/usr/lib/avnav/server/handler/signalkhandler.py", line 1635, in iterateToValue
callback(prefix,node.get('value'),node.get('$source'),node.get('timestamp'))
File "/usr/lib/avnav/server/handler/signalkhandler.py", line 1656, in store
if self.checkOutdated(timestamp):
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/avnav/server/handler/signalkhandler.py", line 1243, in checkOutdated
if timeStamp < oldest:
^^^^^^^^^^^^^^^^^^
TypeError: '<' not supported between instances of 'NoneType' and 'float'

2025-02-23 22:19:25,340-9520-9538-SignalKHandler-INFO-STATUS[main] ERROR, error when fetching from signalk http://localhost:3000/signalk/v1/api/
2025-02-23 22:19:25,343-9520-11523-signalk-websocket-write-INFO-STATUS[write] NMEA, connected
```

Is there anything I have to adapt, to get it working again?

free-x, 2025-02-24:
Is it Real Data? Or Simulated/Recorded Data?

ukoehler, 2025-02-24:
That is real data. As mentioned above, OpenCPN has no problem handling the data.

ukoehler, 2025-02-24:
As far as I understand the error message from the AvNav log, that seem a syntax error:
`TypeError: '<' not supported between instances of 'NoneType' and 'float'`

I extracted all timestamps from the data and they all have the same format. There seem to be no broken one.

wellenvogel, 2025-02-24:
Unfortunately SignalK again changed the format of their JSON responses. There are now strange "value" entries at various points in the hierarchy that do not have a timestamp and also do not contain any meaningful data. From the $source they are created by the new SignalK courseApi.
As AvNav tries to fetch the data from SignalK in a somehow generic way (thus allowing to display all data values) it needs to go through the hierarchy and try to find the really valid data. There is no clear description at the SignalK side what criteria should be used for this approach.
So I will add some small change that will ignore value entries that do not have timestamp or $source. Will become interesting to see what will be broken next time.

ukoehler, 2025-02-24:
Mans thanks for the quick help. Any idea when these changes will be available in OpenPlotter? I can only  see a release from 2015 here in GutHub.

free-x, 2025-02-25:
you have two options
- Either download from daily archive
https://www.wellenvogel.net/software/avnav/downloads/daily/?dir=/software/
wget ....
sudo dpkg -i <package>

- or activate repository for Openplotter in /etc/apt/
````
deb https://www.free-x.de/deb4op bookworm-preview main
````
afterwards
````
sudo apt update
sudo apt upgrade
````

ukoehler, 2025-02-25:
Great! Added your repository and AIS targets are shown again. Many, many thanks for the quick fix!

closed 2025-02-25 by ukoehler