wellenvogel/avnav #406: shorten ap_max_inactivity for the hostapd on the raspberry pi image an fall back for the Wifi firmware on bookworm

wellenvogel, 2025-03-04:
https://github.com/wellenvogel/esp32-nmea2000/issues/102 describes a problem for reconnecting devices to a Pi running bookworm.
On our side there should be 2 changes:
(1) shorten the time for the hostapd to detect a station as inactive (ap_max_inactivity) to 30s
(2) implement the switch to the "minimal" firmware as described in https://github.com/RPi-Distro/firmware-nonfree/issues/49

free-x, 2025-03-04:
Hi @wellenvogel ,

we should also think about the fact that slowly everything in debian/raspios is controlled via NetworkManager. Also WiFi AP

wellenvogel, 2025-03-04:
Agree...

wellenvogel, 2025-07-23:
release with [20250723](https://www.wellenvogel.de/software/avnav/downloads/release/20250723/?lang=en)

closed 2025-07-23 by wellenvogel