wellenvogel/avnav #410: Android Client-server connection on Bluetooth nok

PietKnot, 2025-04-21:
Hi Andreas, long time fully satisfied by AvNav. but now a question as my HW-configuration (very small Android phone) is migrating to "Client - Server" configuration: Android server (tablet) an Android Client (larger phone) but communicating via Bluetooth.

First most simple testconfiguration to start with (not really client-server, but multiplexer orientation):

- tested: Bluetooth serial reception GPS / AIS from external senders by phone or tablet stand-alone oke.
- tested: Bluetooth connection between Tablet (with Internet) and Phone (no internet) tethering using Bluetooth oke.

Thus Bluetooth connection on Internet-sharing oke, available on other protocol: Sharing contacts not tested But COMMUNICATION SERIAL PORTS??

Test configuration at home: tried  Tablet-multiplexer with internal GPS-working sending via Bluetooth connection (active internet sharing) GPS NMEA-data to Phone-receiver NOT working: "no connection can be made".

Question: have you ever tested Android sending NMEA via Bluetooth? It is from the GUI supported but . . . . . working? 
The error string can be given, but a first this simple question as a starter.

wellenvogel, 2025-04-21:
Not sure if I understand correctly what you are trying to do.
For sure bluetooth on AvNav will only work if the device supports the serial profile.
But this is only tested when AvNav is trying to connect.
So if the tethering connection does not support a serial profile it will not work.
But I guess you should be able to use a TCP or UDP connection on this bluetooth connection (as it is basically adding an IP layer).
Not sure if mDNS resolution will work via this connection - but you could try.

PietKnot, 2025-04-23:
Tablet (BT-receiver) and Phone (BT-receiver) are both functioning receiving -home brew- Bluetooth-senders for AIS and GPS.

I'm trying to forward the received NMEA by the tablet to the Phone. In my first test-configuration at home however, i'm using the Tablets internal GPS (working) to send it's position to the Phone via BT. Both are BT-coupled but NOT connected in the test (note: tethering for internet, contact-sync are able to connect using BT!!) . The error presented at both sides is:

"main: failing with connect error java.io.IOException: read socket might closed or timeout, read ret: 01"

Hope the situation is clear now (you could test this at your side also); thus serial profile is supported for receiving info, but SENDING unknown. Please verify, no urgent issue but just a question if BT connected Server-client can work on Android!! Thethering via WiFi is not my intention, was just to verify the BT-connection oke or not.

wellenvogel, 2025-04-24:
No - not Wifi.
Bluetooth tethering...

PietKnot, 2025-04-24:
Maybe i did miss-understand your suggestion: Bluetooth tethering (internet sharing) does work, but have to test if that can be setup without internet connection and be used for ip-communication.

wellenvogel, 2025-04-24:
Exactly.

PietKnot, 2025-04-24:
Can confirm that this is working, although i do not understand the logic fully:

Tablet sender (no SIM-card, not wifi active, no internet) with internat GPS enabled made via Bluetooth a tethering connection to Phone receiver (no WIFI, no internet - data disabled)  with internal GPS disabled.. 

Had a short problem as firewalls were enabled, but disabled them.

Tablet and Phone: MDSNresolver enabled: stating bt-pan; 1 main resolver active

Tablet: TCPwriter enabled gets ip: 192.168.44.1:57484  sending 9.00 / sec.

PietKnot, 2025-04-24:
Did not finish comment:
Tablet: TCPwriter (Test) enabled gets ip: 192.168.44.1:57484 sending 9.00 / sec.
- listening on 444, external access true
- mdns avnav-android.
Phone: NMEA0183 service (Test)
- service avnav-android
- main rcv=9,1/s

I do not understand why TCPreader does not work, but will test UDP sending also. Basics looks fine but would be much easier without tethering  . . . thanks for the idea and support

wellenvogel, 2025-04-24:
Now I'm lost...
Would be helpfull to get a complete picture of the setup and what worked and what not.
With mdns enabled and an service reader there will be no neec for any further setup.

PietKnot, 2025-04-24:
looks like that UDPwriter does not work "avnav-android" cannot be resolved (mnds name entered).

You should not be lost: it's working as i said with TCPwriter and NMEA0183 service active at receiving side, both using "avnav-android" service.

wellenvogel, 2025-04-24:
Udp-writer would need a complete address - not a service name.
On my side mdns resolve on the client does not work.
I will check if anything can be done for that.
This is some universal setup - regardless of the connection type the devices should find each other.

PietKnot, 2025-04-24:
Oke, just tried -out of curiosity- UDPwriter and the (?) stated that in the ipaddress field, i could enter an IP-address to connect, but also a hostname or mnds name like avnav.local. Thus just tried the latter an hour ago, without luck, but i'm happy. Further, i assume that a full client-server connection will cost a lot more more BT-bandwidth than this NMEA connection, thus will stay with this solution for the time being.

For sure, you will manage also to get this "simple" solution up-and-running: no WIFI-router needed, but just build-in BT.

wellenvogel, 2025-04-26:
It turned out that the mDNS resolver was not able to discover services running on newer(?) android devices - regardless of connectivity type. I just made some corrections (see #415). If you are interested you could try out the [daily](https://www.wellenvogel.net/software/avnav/downloads/daily/20250426/).
On the "server" side set up a TCP writer with external access, mdns enabled.
Connect the devices - does not matter which way (i.e. same Wifi router, Wifi tethering, BT tethering).
On the client side use an NMEA0183 service and select the service you have set up on the server side. Set a connect timeout (e.g. 20s) to ensure that a connect will terminate if not successful. 
This way the AvNav instances will always automatically connect and reconnect - even if you change the connectivity on the fly.

PietKnot, 2025-04-28:
Thnxc; will test Daily bit takes some time, sorry.

PietKnot, 2025-05-16:
All works fine with AvNav release version (2024-06-16)  on Android 12 server and Android 10 client. Daily testing not needed anymore.

closed 2025-05-16 by PietKnot

wellenvogel, 2025-05-16:
But just without using mDNS I guess?

PietKnot, 2025-05-18:
no, with mDNS active and working!! Only the UDP-writer did not work as said above, but no reason to keep this issue open. Furthermore possibly Chinese Android version but "Hardware Device Info" app confirms Chinese Andoid version . . . .  but never 100% sure.