wellenvogel/avnav #411: Zoom issue when using multiple charts

ronangaillard, 2025-04-25:
I have two mbtiles charts : one for high zoom levels, one for low levels in a small area. I can't use them both at the same times.

Let's call the high zoom level chart the A chart, and the low level one B chart.

- If I select A chart and add B chart as a layer to it, I can't zoom enough as I guess that the minimum level of the A chart is taken into account.

Example, max zoom level in this configuration :

[image]

And max zoom level if I use B chart alone :

[image]

- If I select B chart and add A chart as a layer to it, A chart will always be in the front so I won't be able to see details in B charts.

I hope I'm clear enough, is this a bug or is there some settings or config that fixes this ?

Thanks !

wellenvogel, 2025-04-26:
You can put layers on top of a chart or below. Just drag and drop in the overlay dialog to change the order.
Most probably this could solve your problem.

ronangaillard, 2025-04-26:
Exact ! Thanks ! So for future reference what works is :

Use most details chart, add layer to it with "global"/unzoomed chart, and drag&drop it above _---chart---_

[image]

wellenvogel, 2025-04-26:
Ok, so the problem is resolved ;-)

closed 2025-04-26 by wellenvogel