wellenvogel/avnav #412: Mbtiles charts are blurry

ronangaillard, 2025-04-25:
(Sorry if I'm posting too many issues)

I am using a mbtiles chart file. When I use it with OpenCPN it's fine, but in avnav it looks blurry. Tried. on Firefox, Chrome, on MacOS and Firefox. Here is a side by side comparison using the exact same file :

[image]

ronangaillard, 2025-04-25:
I also tried j iPhone and I have the same issue 
[image]
[image]

wellenvogel, 2025-04-26:
Can you provide the mbtiles file?

ronangaillard, 2025-04-26:
Thanks for your answer, here is the mbtiles file : https://drive.proton.me/urls/X4VR2QKCH8#0RIGNu3CVM5c

wellenvogel, 2025-04-26:
It's a bit hard to see the problem from your screenshots.
Just consider that if you are on non-int zoom levels the tiles will be scaled up/down from a math rounded zoom level.

ronangaillard, 2025-04-27:
[image]

Here is a side by side comparison with OpenCPN using the exact same source mbtiles file. We can clearly see how it is more blurry in avnav. I did some search on the internet and it may be link with the pixelratio parameter of Open Layers, what do you think ?

I made sure to be on an exact zoom level (13.0) and I still have the issue. It can be so blurry sometimes that it is hard to read text. 

My avnav version : 20240616, and issue is on firefox and chrome.

wellenvogel, 2025-04-27:
Please just use the +/- buttons for the zoom. And have map->"lock to int zoom levels with buttons" active.
My screen shot attached.

[image]

ronangaillard, 2025-04-27:
It indeed seems better thanks ! (But this means we will get blurry maps by zooming with scroll wheel/pinching ? )

Also on the mbtiles it does not work well, please see comparision with OpenCPN :

[image]

After inspecting network request it looks like the issue comes from the tile server as downloaded tile is already blurry :

[image]

Served MBtile vs OpenCPN.

I'm tyring another mbtile server to see if that fixes the issue.

ronangaillard, 2025-04-27:
So I tried with mbtileserver (https://github.com/consbio/mbtileserver) and still same issue :

[image]

There must be some kind of magic happening in OpenCPN, I'll keep investigating.

ronangaillard, 2025-04-27:
After further investigation using QGIS if think that avnav uses tiles that are not zoomed in enough. (Sorry I'm not sure if I'm clear) : instead of showing tiles of zoom X it should show tiles of zoom X+1. that's how OpenCPN and QGIS are configured and why they show a "clear" map.

There is a setting in avnav called `scale the map display` but if I set it to 0 it sayd `out of range`. I think the solution would be to allow 0 or even negative values.

[image]

For example :

with openCPN scale setting of -3 (notice that the setting is inverted compared to avnav) :

[image]

[image]

When changed to 0 it's clear and not blurry :

[image]

[image]

wellenvogel, 2025-04-28:
Maybe to get a hint where we have to search for you could try another map viewer in the browser and compare to OpenCPN (e.g. just OSM).

ronangaillard, 2025-04-28:
I made a test with Leaflet, you can access it here : http://51.38.220.64:3000/

On non integer zoom we can clearly see the difference between avnav and the leaflet example :

[image]

On integer zoom it's a bit less noticeable but still there :

[image]

Avnav : 

[image]

Leaflet :

[image]

wellenvogel, 2025-04-28:
Many thanks for all your effort in testing...
I just repeated your test - and cannot really see a difference.
I attach 2 screenshots - one with ol and AvNav on an int zoom (using the zoom buttons) and one on a not int zoom (using the mouse wheel to zoom).
From my pov both look always the same - either blurry (non int zoom) or non blurry (int zoom).
And this is also the expected behavior. At the end there is not much that is done inside AvNav to display the tiles - it's just left to OpenLayers.
And for non-int zoom levels the library will just pick up a tile of the next matching level and will scale up/down the image.
Non-int zoom:
[image]

Int-zoom:
[image]

wellenvogel, 2025-04-28:
I just played a bit with various options.
In principle it will be possible to set imageSmoothingEnabled to false on the render context. This will creating some aliasing effects but the fonts would be more clear.
Left is current, right with imageSmoothingEnabled==false.
On int zoom levels there is no diff of course.

[image]

This would be supported on all browsers. But I don't think that this would be much better.
Unfortunately imageSmoothingQuality (another option) is not supported on all browsers (especially not on firefox). Setting this to "high" gives better results on Chrome.

wellenvogel, 2025-05-04:
Any further thoughts on this?

ronangaillard, 2025-05-04:
Not that I'm thinking of right now. Thanks for trying on your side !

I'll keep investigating on my side, but I guess for the moment we can close this issue. Maybe I should open one on the Open Layer side. 

Thanks again !

wellenvogel, 2025-05-04:
I don't think that OpenLayers can do anything here. It's just the way how drawImage on a renderingContext2D is working.
Maybe one could get better results using webGl - but this is nothing that could easily be done.
Could you just have a look on my last 2 pictures?
I could create a switch that would allow to turn on some "drawing improvement" - would be the solution I showed for firefox and some real improvement on chrome.
From my pov on firefox that is not really an improvement - and firefox for sure is one of the important browsers.

ronangaillard, 2025-05-04:
Do you have any idea why OpenCPN manages to render tiles so nicely ? Is that because it's using something similar to WebGL like OpenGL ? Yes I would understand that it would be much work... On another side do you think it would improve perfs ? Because if so it would be a great bonus as on reaspberry pi 5, maps in the browser is still a bit laggy. 

I hava had a look at both yout images but I agree with you that it's not much better (maybe even worse?), so I don't think it's worth implementing that switch.

wellenvogel, 2025-05-05:
I just played a bit using the OL WebGL renderer - but from my pov results are not better (I feel even worse for int zoom levels).
Potentially one could improve the shader code to use other scaling algorithms - but for now I will not continue here...

Left ist WebGL, right is the normal canvas 2D drawing.

[image]

Source is based on your code...

```
<!DOCTYPE html>
<html>
<head>
    <title>Ma Carte MBTiles (OpenLayers)</title>
    <meta charset="utf-8" />
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <link rel="stylesheet" href="https://cdn.jsdelivr.net/npm/ol@v9.1.0/ol.css">
    <style>
        html, body {
            height: 100%;
            margin: 0;
            padding: 0;
        }     
        #map {
            height: 100%;
            width: 100%;
        }
        .ol-zd{
            top: 3.5em;
            left: 0.5em;
        }
        .ol-red{
            color:red;
        }
    </style>
</head>
<body>
    <div id="map"></div>
    <script src="https://cdn.jsdelivr.net/npm/ol@v9.1.0/dist/ol.js"></script>
    <script>
        let webgl=false;
        if (window.location.search && window.location.search.match(/webgl=/)){ webgl=true;}
        class ZoomDisplay extends ol.control.Control {
            /**
             * @param {Object} [opt_options] Control options.
             */
            constructor(opt_options) {
                const options = opt_options || {};

                const element = document.createElement('div');
                element.className = 'ol-zd ol-unselectable ol-control';
                if (webgl) element.className+=" ol-red";
                super({
                    element: element,
                    target: options.target,
                });
                this.te=element;
            }
            render(mapEvent){
                let z=this.getMap().getView().getZoom();
                this.te.innerText=z;
            }
        }
        const tileServerUrl = 'http://51.38.220.64:8000/services/07-Granville-Trebeurdin-SHOM/tiles/{z}/{x}/{y}.png';
        const attributionText = '&copy; VosDonnées | Servi par MBTilesServer | &copy; OpenStreetMap contributors';
        const initialCenterLonLat = [-2.01, 49]; 
        const initialZoom = 13;
        const minZoomLevel = 12;
        const maxZoomLevel = 17;

        const tileSource = new ol.source.XYZ({
            url: tileServerUrl, 
            attributions: attributionText,
        });

        let tileLayer;
        if (!webgl) {
                tileLayer = new ol.layer.Tile({
                    source: tileSource,
                });
            }
            else{
                tileLayer = new ol.layer.WebGLTile({
                    source: tileSource
                })
            }

        const mapView = new ol.View({
            center: ol.proj.fromLonLat(initialCenterLonLat),
            zoom: initialZoom,
            minZoom: minZoomLevel,
            maxZoom: maxZoomLevel,
        });
       
        const map = new ol.Map({
            target: 'map',       
            layers: [tileLayer],
            view: mapView,
            controls: [
                new ol.control.Zoom(),
                new ZoomDisplay()
            ]      
        });

        map.addControl(new ol.control.ScaleLine({units: 'metric'}));
        document.querySelector('.ol-zoom-in').onclick=(ev)=>{
            window.setTimeout(()=>{
                let z=map.getView().getZoom();
                z=Math.floor(z);
                map.getView().setZoom(z);
            },500);
            console.log("zoom in");
        }
    </script>
</body>
</html>
```

closed 2025-05-05 by wellenvogel

free-x, 2025-05-07:
@ronangaillard 
Did you try avnav-mapproxy-plugin, that provides also SHOM source?