wellenvogel/avnav #413: Not an issue but an idea belonging to UPD and TCP writer (specially for Android version)

PitHerm, 2025-04-26:
it would be great to add a timer to TCP/UPD writer
from 0 (behavior like normally) up to 60 seconds
to send the newest gps data 
to reduce network traffic and other usage options.

wellenvogel, 2025-04-26:
This is not that simple.
Badically AvNav sends out NNEA data that it receives.
So there would be no easy chance for such a timer.
The only thing that you could control would be the update rate for the internal GPS.
But what would be the use case?
For the intended use (i.e. sail boat navigation) it does not make sense.

wellenvogel, 2025-05-05:
any thoughts?

PitHerm, 2025-05-21:
> any thoughts?

Sorry about the late reply :-(

I understand that problem and created a workaround for my idea to send position depending on interval to my private server.

BTW i spotted a nice menue missmatch within AvNav app latest from playstore:
TCP-Writer <> TCP-Reader

if you choose tcp-writer you only can enter a listener port, if you choose tcp-listener you are eable to ender the send to IP and port . :-)

wellenvogel, 2025-05-21:
> BTW i spotted a nice menue missmatch within AvNav app latest from playstore:
TCP-Writer <> TCP-Reader
if you choose tcp-writer you only can enter a listener port, if you choose tcp-listener you are eable to ender the send to IP and port . :-)

Not really a mismatch. Exactly what the main use case is. The writer is intended to send data to clients that connect to it. The reader is intended to fetch data from other systems.
But at the end they can work bidirectional and you can also use them the other way. Typically I would set up my data flows in a way that reflect the meaning of the system - and this way AvNav is some sort of a central server. But this is up to the user at the end. I could alos name them TCPServer and TCPclient - but not sure if this would make things easier to understand.

For the main part of this issue - I guess I can close it.

closed 2025-05-21 by wellenvogel